nanopb/nanopb #483: Podscpec missing for nanopb 0.3.9.1

mali1989, 2020-01-23:
Hi everyone,

in the following release 0.3.9.1

[https://github.com/nanopb/nanopb/tree/nanopb-0.3.9.1](https://github.com/nanopb/nanopb/tree/nanopb-0.3.9.1)

Could you please add it into the Podspec? We need it for internal referencing purposes

Thanks so much!

PetteriAimonen, 2020-01-23:
I'm sorry, but I have no idea what is a Podspec and who may be maintaining one for nanopb.

mali1989, 2020-01-23:
Here's an excerpt from Cocoapods:

> A Podspec, or Spec, describes a version of a Pod library. One Pod, over the course of time, will have many Specs. It includes details about where the source should be fetched from, what files to use, the build settings to apply, and other general metadata such as its name, version, and description.

PetteriAimonen, 2020-01-23:
So, where is it? I'm not aware of any podspec in nanopb repository.

mali1989, 2020-01-23:
This is what I'm asking - I was looking for such one as I'm used to have it also in other FOSS. That's why I'm asking.

closed 2020-06-21 by PetteriAimonen