nanopb/nanopb #489: 4.1 encoder returns 0

chris1seto, 2020-02-12:
I have a project which has been using nanopb for about a year. I recently upgraded it to 4.1 after the old 3.9 release was returning 0 intermittently after adding a new protobuf field to my message. 

I have the following message defined, and it exists inside of another message, which is then nested again in another message.
```
syntax = "proto3";

message FanConfig
{
  float dutycycle = 1;
  float min_rpm = 2;
  float max_rpm = 3;
}
```
```
syntax = "proto3";
import "FanConfig.proto";

message FansConfig
{
  FanConfig fan_1 = 1;
  FanConfig fan_2 = 2;
  FanConfig fan_3 = 3;
  FanConfig fan_4 = 4;
  FanConfig fan_5 = 5;
  FanConfig fan_6 = 6;
  FanConfig fan_7 = 7;
  FanConfig fan_8 = 8;
  FanConfig fan_9 = 9;
}

```
For each fan_x message, I set `status->fans_status.has_fan_X = true;`. This works fine for the first several messages, but if I enable all of them up to the 6th, the nanopb encoder returns 0. 

Here is my encoding:
```
PbResponse response_msg;
..snip...

    // Deflate the response
    stream_out = pb_ostream_from_buffer(data_tx + HEADER_WORD_SIZE, NET_BUFFER_SIZE - HEADER_WORD_SIZE);
    if (!pb_encode(&stream_out, PbResponse_fields, &response_msg))
    {
      LOG_E(TAG, "Unable to encode");
      netconn_close(conn);
      return;
    }
```

It seems like there must be a bug in nanopb causing the encoder to return 0? Any ideas for how to troubleshoot this?

chris1seto, 2020-02-12:
Nevermind. This was my error in stream buffer size. I'll close now. Sorry for the bad report!

closed 2020-02-12 by chris1seto

PetteriAimonen, 2020-02-13:
Glad to hear you got it fixed!