nanopb/nanopb #491: Use Angle Brackets Instead of Quotes for Include Files

bramankp, 2020-02-15:
In a situation where I want to provide a custom pb.h *outside* the nanopb project source, compiling reverts to the local pb.h because of the use of double quotes.
```
#include "pb.h"
```
should read
```
#include <pb.h>
```

PetteriAimonen, 2020-02-15:
Hmm, why would you want to do that?

In general it doesn't sound like a good idea, because the pb.h definitions are closely tied with the source code, and getting out of sync could cause all kinds of weird bugs.

bramankp, 2020-02-15:
I want to keep the original source untouched so I can see what changes I've made. Any time I update from one version of nanopb to a newer one I expect to update my custom pb.h but this allows me to know what changes I have made to customize it. I'm not using Git submodules to import the code into my project. It's just extracting the source directly, hence the desire *not* to touch the original source code.

PetteriAimonen, 2020-02-15:
Why do you want to modify pb.h? If you just want to alter some defines, you can do that in compiler settings, or using the syshdr mechanism.

bramankp, 2020-02-15:
That's my workaround, yes, but the header does claim I can modify that file directly. :)

There's no sense in negotiating the change if you're against it. I likely won't convince you and it's not a broken feature or anything. Call it "my preference" and maybe I'm in the minority so feel free to close this issue.

PetteriAimonen, 2020-02-15:
Sorry if this sounds harsh, I'm just trying to get to the bottom of the real issue :)

It's true that you won't convince me to make including pb.h from outside the source tree easier. But if the issue is indeed difficulty in modifying the settings, it might be solvable by some other means.

The "you can edit this pb.h" comment was added because people were editing the generated .pb.h files instead, and a `#define` there that's not included in rest of the code caused all kinds of hard-to-debug issues being reported. Not ideal, but best I've figured so far.

bramankp, 2020-02-15:
No worries, I understand your point of view. Thanks for the lively discussion.

closed 2020-02-15 by bramankp

mibcat, 2020-06-19:
Hi!

I'm a little bit confused: 

the generated file **uses** already an `#include <pb.h>`

e.g. 
```bash
~/tmp/github/nanopb/generator/protoc --nanopb_out=. rfm.proto
```
generates the file rfm.pb.h:
```c
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.2-dev */

#ifndef PB_RFM_RFM_PB_H_INCLUDED
#define PB_RFM_RFM_PB_H_INCLUDED
#include <pb.h>

#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif
...

```

Because of this the folder containing these generated files must be in the **compilers' search path** which would not be necessary if quotes were used (and which should be the correct behavior in my point of view).

If I haven't missed a pull request, then I don't understand the generated code ...

Best regards,
Michael.

PetteriAimonen, 2020-06-20:
@mibcat The generated files use `#include <pb.h>` because in general, they are not in the same directory as nanopb source. On the other hand, nanopb source files such as `pb_encode.c` use `#include "pb.h"`, because they are in the same directory as nanopb source.

mibcat, 2020-06-20:
Thanks for your feedback !

I'm using Eclipse CDT managed build and my project structure looks like:
```bash
src
├── main.cpp
├── protobuf
│   ├── pb_common.c
│   ├── pb_common.h
│   ├── pb_decode.c
│   ├── pb_decode.h
│   ├── pb_encode.c
│   ├── pb_encode.h
│   ├── pb.h
│   ├── README.md
│   ├── rfm.pb.c
│   ├── rfm.pb.h
│   └── rfm.proto
```

The generated protobuf  header `rfm.pb.h` is included in main.cpp by:
```c++
#include <protobuf/rfm.pb.h>
```

With the original generated code (using angle brackets) the project build fails because protobuf is not added explicitly to the compilers' search path - only 3 additional paths -I (shortened with ...): 
```bash
Building file: ../src/protobuf/rfm.pb.c
Invoking: Cross GCC Compiler
arm-linux-gnueabihf-gcc 
-std=c11 -DRH_PLATFORM=RH_PLATFORM_RASPI_PIGPIOD 
-I"/hom ... aspi/src" 
-I"/hom ... aspi/cross_compile/arm-linux-gnueabihf-output/include" 
-I"/hom ... aspi/src/RadioHead" 
-O0 -g3 -Wall -Wextra -c -fmessage-length=0 -MMD -MP -MF"src/protobuf/rfm.pb.d" -MT"src/protobuf/rfm.pb.o" -o "src/protobuf/rfm.pb.o" "../src/protobuf/rfm.pb.c"
In file included from ../src/protobuf/rfm.pb.c:4:0:
../src/protobuf/rfm.pb.h:6:16: fatal error: pb.h: No such file or directory
src/protobuf/subdir.mk:27: recipe for target 'src/protobuf/rfm.pb.o' failed
 #include <pb.h>
                ^
compilation terminated.
make: *** [src/protobuf/rfm.pb.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.
```

If changing the include to quotes `#include <pb.h>` the path protobuf must **not** be added to the compilers' search path to get a successful build:
```bash
Building file: ../src/protobuf/rfm.pb.c
Invoking: Cross GCC Compiler
arm-linux-gnueabihf-gcc 
-std=c11 -DRH_PLATFORM=RH_PLATFORM_RASPI_PIGPIOD 
-I"/hom ... aspi/src" 
-I"/hom ... aspi/cross_compile/arm-linux-gnueabihf-output/include" 
-I"/hom ... aspi/src/RadioHead" 
-O0 -g3 -Wall -Wextra -c -fmessage-length=0 -MMD -MP -MF"src/protobuf/rfm.pb.d" -MT"src/protobuf/rfm.pb.o" -o "src/protobuf/rfm.pb.o" "../src/protobuf/rfm.pb.c"

Building file: ../src/main.cpp
Invoking: Cross G++ Compiler
arm-linux-gnueabihf-g++ 
-std=c++1y -DRH_PLATFORM=RH_PLATFORM_RASPI_PIGPIOD 
-I"/hom ... aspi/src" 
-I"/hom ... aspi/cross_compile/arm-linux-gnueabihf-output/include" 
-I"/hom ... aspi/src/RadioHead" 
-O0 -g3 -Wall -Wextra -c -fmessage-length=0 -MMD -MP -MF"src/main.d" -MT"src/main.o" -o "src/main.o" "../src/main.cpp"
Finished building: ../src/protobuf/rfm.pb.c
```

How could this behavior be explained ??

PetteriAimonen, 2020-06-20:
Sounds like normal behavior for cpp.

If you want to have quotes in the generated files, use the `--library-include-format` option:

    python generator/nanopb_generator.py --library-include-format='#include "%s"' foobar.proto

mibcat, 2020-06-20:
Thanks - didn't know this option. This is exactly what I need :smiley: