nanopb/nanopb #493: Maximum recursion depth exceeded

sbosse-esmart, 2020-02-18:
I did a dumb error today. I nested a child message in a `oneof` with the same name of its parent which causes a runtime error in  `nanopb_generator.py (maximum recursion depth exceeded)`.

Not sure if it's the role of `protoc` or `nanopb` to parse this kind of error, but it could be nice to add. Here is the faulty code : 

```protobuf
message MyMessage {
    option (nanopb_msgopt).msgid = 8;
    oneof configuration_flash {
        MyMessage my_message_v1 = 1;
    }
}
```

PetteriAimonen, 2020-02-19:
Hmm yeah, in callback mode that should work and in non-callback mode it should provide a more meaningful error, so definitely some kind of bug.

cajun-rat, 2020-03-16:
Just a note for the next person who finds this issue: it is possible to work-around this by forcing the field type to `FT_CALLBACK`  See the [cyclic_messages](https://github.com/nanopb/nanopb/tree/master/tests/cyclic_messages) test for an example. The fix here is about auto-detecting this case.

PetteriAimonen, 2020-11-25:
I fixed the recursion depth error, but it will still fail at compilation time if there is a cyclic message. But hopefully that is at least somewhat clearer.

It's not that trivial to autodetect recursive messages and especially it is not clear at which point the recursion should be broken, so for now the generator does not try to automatically set FT_CALLBACK for them.

closed 2020-11-25 by PetteriAimonen