nanopb/nanopb #494: Using sizeof on anonymous union not allowed in C++

torfinnberset, 2020-02-25:
The code generated by https://github.com/nanopb/nanopb/blob/1bcc1230f50482c8d0d7eaf23431d884eeb2b3eb/generator/nanopb_generator.py#L927 creates invalid C++ code (but valid C code). Here is an example I distilled down from some generated nanopb code 

```C++
#define SubMessageSize 20

int max_size = (11 + sizeof(union {
    char f0[70];
    char f2[SubMessageSize];
}));
``` 
which you can see on godbolt here: https://godbolt.org/z/-fV52C.

To create valid C++ code, we have to first give the union a name, which has to be done in a separate statement. I made a hack to show a potential solution:

```diff
@@ -1449,8 +1449,15 @@ class ProtoFile:
             for msg in self.messages:
                 msize = msg.encoded_size(self.dependencies)
                 identifier = '%s_size' % msg.name
+                
                 if msize is not None:
-                    yield '#define %-40s %s\n' % (identifier, msize)
+                    if 'union' in str(msize):
+                        union_name = F"{identifier}_size_union"
+                        union_data = re.match(R'.+sizeof\(union{(.+)}\).+', str(msize))[1]
+                        yield 'union %s{%s};\n' % (union_name, union_data)
+                        yield '#define %-40s sizeof(%s)\n' % (identifier, union_name)
+                    else:
+                        yield '#define %-40s %s\n' % (identifier, msize)
                 else:
                     yield '/* %s depends on runtime parameters */\n' % identifier
             yield '\n'
```

but perhaps you know a cleaner way to inject the union.

PetteriAimonen, 2020-02-25:
Earlier report of the same: #415.

The workaround approach seems ok, though it will be better to inline the change into where `union {}` is generated instead of regexping it after the fact.

But overall I'll still rank this as a rather low priority, as it is better in any case to have the dependencies visible to nanopb generator at generation time. This may need fiddling a bit with `-I`  paths, depending on the directory structure, but should be doable.

torfinnberset, 2020-02-25:
I agree making the change where the `union` is generated would be better, but it's currently not possible as it's a sub-expression (in my example `11 + sizeof...`), so I think that would require significant refactoring. 

If the issue can also be solved by making the dependencies visible at generator then I'd focus on solving that instead. If you find it useful, I could clean up my patch for a PR though.

PetteriAimonen, 2020-02-25:
Having the dependencies visible has been a long-standing problem. It works in most cases, but then again seems to break in others. Solving it would be great for things other than just the size macros, knowing the sizes more accurately in generation time also allows more accurate sizing of the descriptor arrays and avoids problems with enum signedness.

How are you currently invoking nanopb generator and with what kind of .proto when you get that problem? Does using `python nanopb_generator.py foo.proto bar.proto` syntax make a difference?

PetteriAimonen, 2020-02-25:
And yeah, go ahead and pull request that patch. Indeed the refactoring to avoid the regexp could be quite large, and even as it is the hack is local enough that it shouldn't cause much of trouble.

torfinnberset, 2020-02-25:
I tried to narrow down the scenario where this code is generated, and it seems it happens when I include a message from another _protobuf package_ in the `oneof`. Otherwise it seems to work, even when the messages are in multiple files. Does this make sense? 

So, this would not work if `AnotherMessage` is in a different package:
```proto
message OneofAKindTest {
  oneof the_first {
    fixed32 a_fixed_integer = 1;
    other_package.AnotherMessage a_message = 2;
  }
}
```

PetteriAimonen, 2020-02-25:
Yeah, this makes sense. Must be some kind of bug in the dependency name resolution.

torfinnberset, 2020-02-25:
Ok, I'll still make the patch by the end of this week so that at least there's a working fallback for the remaining cases.

kjkjava, 2020-04-20:
I'm running into this problem when I `oneof` a message defined in another file, either directly or transitively. Let me know if any workarounds or patches are available!

PetteriAimonen, 2020-04-21:
@kjkjava Try passing all `.proto` files to the generator at once. That should work, unless you are being hit by the multiple packages bug.

kjkjava, 2020-04-21:
Thanks, @PetteriAimonen. This may be difficult due to the way it's integrated into our build system, but I'll look into it.

closed 2020-06-23 by PetteriAimonen