nanopb/nanopb #498: cppcheck warnings

jkmnt, 2020-03-10:
Hi.

I'm using the nanopb (great library!) in our greenhouses automation project.
Code base is checked by cppcheck static code analizer (http://cppcheck.sourceforge.net/)

Nanopb is now triggering some warnings. Most likely a false positives, but I think it worth 
to take a look anyway.

cppcheck version is 1.90,  flags are --std=c99 --enable=warning --inconclusive
nanopb version 0.4.1 (though 0.3.9 triggers similar warnings too)

here is the offending output:

`pb\pb_decode.c:1146:41: error: Array 'fields_seen.bitfield[2]' accessed at index 2, which is out of bounds. [arrayIndexOutOfBounds]

                if (fields_seen.bitfield[req_field_count >> 5] !=

                                        ^

pb\pb_decode.c:1132:31: note: Assignment 'req_field_count=64', assigned value is 64

            req_field_count = PB_MAX_REQUIRED_FIELDS;

                              ^

pb\pb_decode.c:1146:41: note: Array index out of bounds

                if (fields_seen.bitfield[req_field_count >> 5] !=

                                        ^

pb\pb_decode.c:1147:30: error: Shifting 32-bit value by 32 bits is undefined behaviour [shiftTooManyBits]

                    (allbits >> (uint_least8_t)(32 - (req_field_count & 31))))

                             ^

pb\pb_decode.c:1132:31: note: Assignment 'req_field_count=64', assigned value is 64

            req_field_count = PB_MAX_REQUIRED_FIELDS;

                              ^

pb\pb_decode.c:1147:30: note: Shift

                    (allbits >> (uint_least8_t)(32 - (req_field_count & 31))))

                             ^

pb\pb_encode.c:59:19: warning: Possible null pointer dereference: buf [nullPointer]

        dest[i] = buf[i];

                  ^

pb\pb_encode.c:172:37: note: Calling function 'pb_write', 2nd argument 'NULL' value is 0

            return pb_write(stream, NULL, size); /* Just sizing.. */

                                    ^

pb\pb_encode.c:89:32: note: Calling function 'buf_write', 2nd argument 'buf' value is 0

        if (!buf_write(stream, buf, count))

                               ^

pb\pb_encode.c:59:19: note: Null pointer dereference

        dest[i] = buf[i];

                  ^

pb\pb_decode.c:583:5: error: Common realloc mistake: 'ptr' nulled but not freed upon failure [memleakOnRealloc]

    ptr = pb_realloc(ptr, array_size * data_size);

    ^ `

PetteriAimonen, 2020-03-11:
Hmm yeah, some of these are false positives like the "Common realloc mistake", as there is an `if` for exactly that right after it.

But the warnings about `PB_MAX_REQUIRED_FIELDS` are actually valid if `PB_MAX_REQUIRED_FIELDS` is changed from its default to a value that is not a multiple of 32.

It would probably be useful to incorporate cppcheck into the regular test set like splint is, and add exceptions for manually checked false positives.

andrealima, 2020-08-03:
Hi, @PetteriAimonen you intend clean these warnings and error from the code?
If yes, when we could wait for a new version?

ps.: I detected these same warnings/error on version 0.4.2

PetteriAimonen, 2020-08-04:
@andrealima Currently I have no timeframe for this.