nanopb/nanopb #501: problem:how to generate pb_msgdesc_t array

snikeguo, 2020-03-18:
I am developing an small RPC (Embedded MCU) framework:

C# is the IDL language. (IDL file - > compile - > reflect to get the field, so I don't need to write the parser code.  and I intend to support uint8_t /uint8_t array   type )

1.how to generate pb_msgdesc_t array?

2.For non-fixed-length arrays, I want to generate pointer variables. Only using "malloc"  when decode.  (when encoding, without the malloc)

Attachment:
C# IDL：
```
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using EmbedXrpcIdlParser;

enum Sex_t:byte
{
    Man,
    Woman
}
struct Result_t
{
    [FieldIndex(0)]
    int Value;

    [FieldIndex(1)]
    byte NameLen;

    [FieldIndex(2)]
    [MaxCount(IsFixed=true, MaxCount =32,LenFieldIndex =1)]
    byte[] Name;

    [FieldIndex(3)]
    sbyte Arg1;
}
struct  Student_t
{
    [FieldIndex(0)]
    UInt16 ResultsLen2;

    [FieldIndex(1)]
    [MaxCount(IsFixed = true, MaxCount = 32, LenFieldIndex = 0)]
    Result_t[] Results;

    [FieldIndex(2)]
    int Age;

    [FieldIndex(3)]
    byte NameLen;

    [FieldIndex(4)]
    [MaxCount(IsFixed = false, MaxCount = 64, LenFieldIndex = 3)]
    byte[] Name;

    [FieldIndex(5)]
    byte StudentIdLen;

    [FieldIndex(6)]
    [MaxCount(IsFixed = true, MaxCount = 100, LenFieldIndex = 5)]
    byte[] StudentId;

    [FieldIndex(7)]
    Sex_t Sex;
}
struct DateTime_t
{
    [FieldIndex(0)]
    int Year;
    [FieldIndex(1)]
    int Month;
    [FieldIndex(2)]
    int Day;
    [FieldIndex(3)]
    int Hour;
    [FieldIndex(4)]
    int Min;
    [FieldIndex(5)]
    int Sec;
}
struct StudentArray_t
{
    [FieldIndex(0)]
    byte StudentIdLen;

    [FieldIndex(1)]
    [MaxCount(IsFixed = true, MaxCount = 64, LenFieldIndex = 0)]
    Student_t[] Students;
}
delegate void BroadcastDataTime([FieldIndex(0)]DateTime_t t);
interface IMyInterface
{
        Student_t GetStudentInfoFormStudentId(
        [FieldIndex(0)]
        byte StudentIdLen,

        [FieldIndex(1)]
        [MaxCount(IsFixed = true, MaxCount = 100,LenFieldIndex = 1)]
        byte[] StudentId,

        [FieldIndex(1)]
        int arg2,

        [FieldIndex(2)]
        int arg3
        );

    StudentArray_t GetStudentsInfoFormAge(
        [FieldIndex(0)]int Age
        );//客户端从服务器获取数据：根据姓名筛选出那个学生
}

[GenerationOptionParameter(OutPutFileName="StudentService")]
public struct GenerationOption
{

}

```

code gen:
//StudentService.h
```
#include "EmbedXrpcBaseType.h"
typedef enum _Sex_t
{
    Man = 0,
    Woman = 1,
} Sex_t;
typedef struct _Result_t
{
    Int32 Value;
    Byte NameLen;
    Byte Name[32];
    SByte Arg1;
} Result_t;
typedef struct _Student_t
{
    UInt16 ResultsLen2;
    Result_t Results[32];
    Int32 Age;
    Byte NameLen;
    Byte *Name;
    Byte StudentIdLen;
    Byte StudentId[100];
    Sex_t Sex;
} Student_t;
typedef struct _DateTime_t
{
    Int32 Year;
    Int32 Month;
    Int32 Day;
    Int32 Hour;
    Int32 Min;
    Int32 Sec;
} DateTime_t;
typedef struct _StudentArray_t
{
    Byte StudentIdLen;
    Student_t Students[64];
} StudentArray_t;
typedef void (*BroadcastDataTime)(DateTime_t t);
class IMyInterfaceClientImpl
{
public:
    EmbedXrpcClientObject *RpcClientObject = nullptr;
    IMyInterfaceClientImpl(EmbedXrpcClientObject *rpcobj)
    {
        this->RpcClientObject = rpcobj;
        RpcClientObject->ServicesName.Add("IMyInterface.GetStudentInfoFormStudentId");
        RpcClientObject->ServicesName.Add("IMyInterface.GetStudentsInfoFormAge");
    }
    Option<Student_t> GetStudentInfoFormStudentId(Byte StudentIdLen, Byte StudentId[100], Int32 arg2, Int32 arg3)
    {
        **//write serialization code:GetStudentInfoFormStudentId(StudentIdLen,StudentId,arg2,arg3,)**
    }
    Option<StudentArray_t> GetStudentsInfoFormAge(Int32 Age)
    {
        **//write serialization code:GetStudentsInfoFormAge(Age,)**
    }
};

```

snikeguo, 2020-03-18:
https://github.com/snikeguo/EmbedXrpc

PetteriAimonen, 2020-03-18:
You'll need to generate a `.proto` description of your protocol, and then run it through nanopb_generator.py.

closed 2020-03-21 by snikeguo