nanopb/nanopb #506: find_package broken with cmake 3.17

twam, 2020-03-27:
We're currently using cmake's find_package to include nanopb in our project. Currently this is done using `find_package(Nanopb REQUIRED)`.
Starting with cmake 3.17 this throws the following warning:
```
CMake Warning (dev) at /usr/local/Cellar/cmake/3.17.0_1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (NANOPB)
  does not match the name of the calling package (Nanopb).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  external/nanopb/extra/FindNanopb.cmake:341 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  xxx:27 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

In FindNanopb.cmake the find_package_handle_standard_args is currently called 
```
FIND_PACKAGE_HANDLE_STANDARD_ARGS(NANOPB DEFAULT_MSG
...
```
with NANOPB in uppercase. However if I change my include to
`find_package(NANOPB REQUIRED)`
it cannot find the Find*.cmake file anymore:
```
CMake Error at xxx.cmake:27 (find_package):
  By not providing "FindNANOPB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "NANOPB", but
  CMake did not find one.

  Could not find a package configuration file provided by "NANOPB" with any
  of the following names:

    NANOPBConfig.cmake
    nanopb-config.cmake

  Add the installation prefix of "NANOPB" to CMAKE_PREFIX_PATH or set
  "NANOPB_DIR" to a directory containing one of the above files.  If "NANOPB"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:27 (include)
  ```

So I guess either the file needs to be renamed to FindNANOPB.cmake or we must use Nanopb when calling find_package_handle_standard_args.

PetteriAimonen, 2020-03-27:
Hmm, this is beyond my cmake understanding.

twam, 2020-03-27:
You should decide for one spelling of nanopb: `nanopb`, `Nanopb`, `NANOPB` and then we use it all the times :)

PetteriAimonen, 2020-03-27:
Ok :)

So uh, which would cause least problems for existing users?

twam, 2020-03-27:
If we go with `NANOPB`:
* Users need to chance from `find_package(Nanopb REQUIRED)` to `find_package(NANOPB REQUIRED)`
* We need to rename the file to `FindNANOPB.cmake`
If we go with `Nanopb`:
* Users should for `Nanopb_FOUND` instead of `NANOPB_FOUND`, but looks like `NANOPB_FOUND` will be also set for backwards compatiblity (see https://cmake.org/cmake/help/latest/module/FindPackageHandleStandardArgs.html)
* As all other variabes, e.g. NANOPB_SRCS are set manually, they still would be available as before (but kinda inconsistent)
`nanopb` is the worst of both. :) 

Therefor `Nanopb` is most likely the best option in terms of backward compatiblity. I can create PR if it helps.

PetteriAimonen, 2020-03-27:
That sounds good, PR is welcome :)

closed 2020-06-23 by PetteriAimonen