nanopb/nanopb #511: Undefined reference to #MessageName#_msg

OscarVanL, 2020-04-04:
I'm experiencing an issue trying to use nanopb on my Arduino Nano.

When I include the line `bool status = pb_encode(&stream, GpsPayload_fields, &payload);`

I get the compiation error:

```
C:\Users\-----\AppData\Local\Temp\cck7G7g6.ltrans0.ltrans.o: In function `loop':

C:\Users\-----\Documents\GitHub\COMP3210-Bike-Tracker\Bike_Tracker/Bike_Tracker.ino:212: undefined reference to `GpsPayload_msg'

C:\Users\-----\Documents\GitHub\COMP3210-Bike-Tracker\Bike_Tracker/Bike_Tracker.ino:212: undefined reference to `GpsPayload_msg'

collect2.exe: error: ld returned 1 exit status
```

I've been following the guide [here](https://www.dfrobot.com/blog-1161.html).

My nanopb version is `nanopb-0.4.1-windows-x86`

Here's my schema:

```
syntax = "proto2";

message GpsPayload {
	required float latitude = 1;
	required float longitude = 2;
	required int64 epoch = 3;
	required int32 battery = 4;
	optional float speed_kph = 5;
	optional float heading = 6;
	optional float altitude = 7;
}
```

Here's a snippet of the relevant parts of my code.

```
#include "protobuf/BikeTrackerPayload.pb.h"
#include "pb_common.h"
#include "pb.h"
#include "pb_encode.h"

uint8_t buffer[128];  // Todo: resize
GpsPayload payload = GpsPayload_init_zero;
pb_ostream_t stream = pb_ostream_from_buffer(buffer, sizeof(buffer));
payload.latitude = latitude;
payload.longitude = longitude;
payload.epoch = now();
uint16_t batt_percent;
if (fona.getBattPercent(&batt_percent)) {
  payload.battery = batt_percent;
}
payload.speed_kph = speed_kph;
payload.heading = heading;
payload.altitude = altitude;
// Serialise Protocol Buffer
bool status = pb_encode(&stream, GpsPayload_fields, &payload);
```

Any suggestions?

PetteriAimonen, 2020-04-04:
It sounds like `BikeTrackerPayload.pb.c` is not getting included into your project for some reason.

OscarVanL, 2020-04-04:
OK here's where it went wrong.

I followed that guide, which had me use `#include "proto_files/BikeTrackerPayload.pb.h"`, changing this to `#include "proto_files/BikeTrackerPayload.pb.c"` fixed the problem.

After changing this, it complained that it couldn't find BikeTrackerPayload.pb.h, but this is because the import inside BikeTrackerPayload.pb.c did not assume the .h and .c files were in the same directory. Changing `#include "proto_files/BikeTrackerPayload.pb.h"` to `#include "BikeTrackerPayload.pb.h"` fixed this.

My code now compiles without issue and this error no longer occurs.

closed 2020-04-04 by OscarVanL