nanopb/nanopb #514: How to use PB_ENABLE_MALLOC? Oneof still allocates max size.

Imara90, 2020-04-08:
Hi everyone!

So I'm working on a nested Oneof message. I designed a wrapper that should contain submodule messages. It can contain simple log_messages, but it can also contain large camera data byte arrays (in IO). I've now came across the issue that creating a wrapper with a Oneof variable always allocates memory on the stack for the max size (which is a camera capture in my case). This causes some of my log messages to create a stack overflow. I want to use PB_ENABLE_MALLOC to dynamically allocate the sizes, but I cannot find an example for it. If I define PB_ENABLE_MALLOC in pb.h by removing the comment marking, the issue remains. 

Also, what do the ;'s stand for in the size definition of my generated wrapper source code?

Thanks in advance!

```C
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.1 */

#ifndef PB_WRAPPER_PB_H_INCLUDED
#define PB_WRAPPER_PB_H_INCLUDED
#include <pb.h>
#include "debug_commands.pb.h"
#include "log_messages.pb.h"
#include "io.pb.h"
#include "memory.pb.h"
#include "commands.pb.h"
#include "connectivity.pb.h"

#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif

#ifdef __cplusplus
extern "C" {
#endif

/* Enum definitions */
typedef enum _Wrapper_Type {
    Wrapper_Type_REQUEST = 0,
    Wrapper_Type_RESPONSE = 1
} Wrapper_Type;

/* Struct definitions */
typedef struct _Wrapper {
    int32_t start;
    int32_t protocol_version;
    Wrapper_Type type;
    int32_t total_packages;
    int32_t package_index;
    pb_callback_t cb_payload;
    pb_size_t which_payload;
    union {
        DebugCommand debug_command;
        LogMessage log_message;
        IO io;
    } payload;
    bool error;
    int32_t end;
} Wrapper;

/* Helper constants for enums */
#define _Wrapper_Type_MIN Wrapper_Type_REQUEST
#define _Wrapper_Type_MAX Wrapper_Type_RESPONSE
#define _Wrapper_Type_ARRAYSIZE ((Wrapper_Type)(Wrapper_Type_RESPONSE+1))

/* Initializer values for message structs */
#define Wrapper_init_default                     {0, 0, _Wrapper_Type_MIN, 0, 0, {{NULL}, NULL}, 0, {DebugCommand_init_default}, 0, 0}
#define Wrapper_init_zero                        {0, 0, _Wrapper_Type_MIN, 0, 0, {{NULL}, NULL}, 0, {DebugCommand_init_zero}, 0, 0}

/* Field tags (for use in manual encoding/decoding) */
#define Wrapper_debug_command_tag                6
#define Wrapper_log_message_tag                  7
#define Wrapper_io_tag                           8
#define Wrapper_start_tag                        1
#define Wrapper_protocol_version_tag             2
#define Wrapper_type_tag                         3
#define Wrapper_total_packages_tag               4
#define Wrapper_package_index_tag                5
#define Wrapper_error_tag                        14
#define Wrapper_end_tag                          15

/* Struct field encoding specification for nanopb */
#define Wrapper_FIELDLIST(X, a) \
X(a, STATIC,   SINGULAR, INT32,    start,             1) \
X(a, STATIC,   SINGULAR, INT32,    protocol_version,   2) \
X(a, STATIC,   SINGULAR, UENUM,    type,              3) \
X(a, STATIC,   SINGULAR, INT32,    total_packages,    4) \
X(a, STATIC,   SINGULAR, INT32,    package_index,     5) \
X(a, STATIC,   ONEOF,    MSG_W_CB, (payload,debug_command,payload.debug_command),   6) \
X(a, STATIC,   ONEOF,    MSG_W_CB, (payload,log_message,payload.log_message),   7) \
X(a, STATIC,   ONEOF,    MSG_W_CB, (payload,io,payload.io),   8) \
X(a, STATIC,   SINGULAR, BOOL,     error,            14) \
X(a, STATIC,   SINGULAR, INT32,    end,              15)
#define Wrapper_CALLBACK NULL
#define Wrapper_DEFAULT NULL
#define Wrapper_payload_debug_command_MSGTYPE DebugCommand
#define Wrapper_payload_log_message_MSGTYPE LogMessage
#define Wrapper_payload_io_MSGTYPE IO

extern const pb_msgdesc_t Wrapper_msg;

/* Defines for backwards compatibility with code written before nanopb-0.4.0 */
#define Wrapper_fields &Wrapper_msg

/* Maximum encoded size of messages (where known) */
#define Wrapper_size                             (64 + sizeof(union{char f0[1033]; char f6[DebugCommand_size];}))

#ifdef __cplusplus
} /* extern "C" */
#endif

#endif
```

PetteriAimonen, 2020-04-08:
You could either just do `Wrapper *mymsg = malloc(sizeof(Wrapper));` in your own code to allocate the whole message on the heap, or use `[(nanopb).type = FT_POINTER]` option on the oneof fields to make nanopb individually allocate them on the heap.

Imara90, 2020-04-08:
Ok. I've added it to the options file. Then, all thats needed is to also initialize a submessage and assign the pointer in the wrapper? No actual mallocs right? Or can I just start by assigning values to the pointers to the submodules as nanopb handles the allocation?

Imara90, 2020-04-08:
I've made a small change to optimize for file dependencies. The wrapper includes a submodule message that contains a oneof of the potential payloads. 
```C
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.1 */

#ifndef PB_WRAPPER_PB_H_INCLUDED
#define PB_WRAPPER_PB_H_INCLUDED
#include <pb.h>
#include "wrapper_payload.pb.h"

#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif

#ifdef __cplusplus
extern "C" {
#endif

/* Enum definitions */
typedef enum _Wrapper_Type {
    Wrapper_Type_REQUEST = 0,
    Wrapper_Type_RESPONSE = 1
} Wrapper_Type;

/* Struct definitions */
typedef struct _Wrapper {
    int32_t start;
    int32_t protocol_version;
    Wrapper_Type type;
    int32_t total_packages;
    int32_t package_index;
    pb_callback_t cb_submodule;
    bool has_submodule;
    Payload submodule;
    bool error;
    int32_t end;
} Wrapper;

/* Helper constants for enums */
#define _Wrapper_Type_MIN Wrapper_Type_REQUEST
#define _Wrapper_Type_MAX Wrapper_Type_RESPONSE
#define _Wrapper_Type_ARRAYSIZE ((Wrapper_Type)(Wrapper_Type_RESPONSE+1))

/* Initializer values for message structs */
#define Wrapper_init_default                     {0, 0, _Wrapper_Type_MIN, 0, 0, {{NULL}, NULL}, false, Payload_init_default, 0, 0}
#define Wrapper_init_zero                        {0, 0, _Wrapper_Type_MIN, 0, 0, {{NULL}, NULL}, false, Payload_init_zero, 0, 0}

/* Field tags (for use in manual encoding/decoding) */
#define Wrapper_start_tag                        1
#define Wrapper_protocol_version_tag             2
#define Wrapper_type_tag                         3
#define Wrapper_total_packages_tag               4
#define Wrapper_package_index_tag                5
#define Wrapper_submodule_tag                    6
#define Wrapper_error_tag                        14
#define Wrapper_end_tag                          15

/* Struct field encoding specification for nanopb */
#define Wrapper_FIELDLIST(X, a) \
X(a, STATIC,   SINGULAR, INT32,    start,             1) \
X(a, STATIC,   SINGULAR, INT32,    protocol_version,   2) \
X(a, STATIC,   SINGULAR, UENUM,    type,              3) \
X(a, STATIC,   SINGULAR, INT32,    total_packages,    4) \
X(a, STATIC,   SINGULAR, INT32,    package_index,     5) \
X(a, STATIC,   OPTIONAL, MSG_W_CB, submodule,         6) \
X(a, STATIC,   SINGULAR, BOOL,     error,            14) \
X(a, STATIC,   SINGULAR, INT32,    end,              15)
#define Wrapper_CALLBACK NULL
#define Wrapper_DEFAULT NULL
#define Wrapper_submodule_MSGTYPE Payload
[... 92 lines cut ...]
    // Submodule setup
    msg.has_submodule = true;
    msg.submodule.which_payload = Payload_log_message_tag;

    msg.submodule.payload.log_message->which_log = LogMessage_generic_message_tag;
    msg.submodule.payload.log_message->log.generic_message.type = level;
    memcpy(msg.submodule.payload.log_message->log.generic_message.message,
            char_msg, strlen(char_msg));
```
or like this:
```C
    Wrapper msg = Wrapper_init_zero;
    msg.start = MAGIC_START;
    msg.end = MAGIC_END;
    msg.protocol_version = PROTOCOL_VERSION;
    // Submodule setup
    msg.has_submodule = true;
    msg.submodule.which_payload = Payload_log_message_tag;

    LogMessage log = LogMessage_init_zero;
    log.which_log = LogMessage_generic_message_tag;
    log.log.generic_message.type = level;
    memcpy(log.log.generic_message.message, char_msg, strlen(char_msg));
    msg.submodule.payload.log_message = &log;
```

Imara90, 2020-04-08:
I think I fixed it by doing the following:

```C
static void log_puts(LogMessage* log)
{
    // Wrapper setup
    Wrapper msg = Wrapper_init_zero;
    msg.start = MAGIC_START;
    msg.end = MAGIC_END;
    msg.protocol_version = PROTOCOL_VERSION;
    // Submodule setup
    msg.has_submodule = true;
    msg.submodule.which_payload = Payload_log_message_tag;
    msg.submodule.payload.log_message = log;

    // encode
    size_t message_length;
    bool status = 0;
    pb_ostream_t stream = pb_ostream_from_buffer(g_protocol_buffer,
                                                sizeof(g_protocol_buffer));
    status = pb_encode(&stream, Wrapper_fields, &msg);
    message_length = stream.bytes_written;

    send_serial_framed_protobuf(message_length, &LOGGING_UART);
}
```
As the submodule is now a seperate message with the `type:FT_POINTER` option. 
Is this the preferred method? Or am i missing an allocation somewhere now? I am receiving my logmessages ok.

syed00727, 2020-04-10:
@Imara90  could you please help me, I want to decode EnergyTransactionRequest message from this struct. How can I do it ? 
```
typedef struct _MeshFrame {
    uint64_t time;
    pb_size_t which_payload;
    union {
        NodeControl control;
        EnergyState state;
        EnergyTransactionRequest transaction;
        EnergyTransactionStatus transactionStatus;
        HardwareConfig hw;
    } payload;
/* @@protoc_insertion_point(struct:MeshFrame) */
} MeshFrame;

```

I Tried this decoding function but getting error on :  `error: 'MeshFrame {aka struct _MeshFrame}' has no member named 'cb_payload' Mframe.cb_payload.funcs.decode = Mframe_callback;`

```

bool Mframe_callback(pb_istream_t *stream, const pb_field_t *field, void **arg)
{
    MeshFrame *topmsg = field->message;
    printf("time: %d\n", (int)topmsg->time);
    if (field->tag == MeshFrame_transaction_tag)
    {
        EnergyTransactionRequest *msg = field->pData;
    }
    return true;
}
```

    MeshFrame Mframe = MeshFrame_init_default;
    Mframe.cb_payload.funcs.decode = Mframe_callback;
    if (!pb_decode_nullterminated(&stream, MeshFrame_fields, &Mframe))
    {
        fprintf(stderr, "Decoding failed: %s\n", PB_GET_ERROR(&stream));
    }
    if (Mframe.which_payload == MeshFrame_transaction_tag)
    {
        MDF_LOGD("MeshFrame_transaction_tag");
    }

Imara90, 2020-04-14:
@syed00727, I don't think that's a related issue at all, but I think you haven't added the option to enable callbacks. 

You should add `option (nanopb_msgopt).submsg_callback = true;` to the MeshFrame message. Also be sure to import nanopb at the top of your file. `import "nanopb.proto";`.

PetteriAimonen, 2020-06-21:
If I understood correctly, you got the problem resolved.

closed 2020-06-21 by PetteriAimonen