nanopb/nanopb #517: Optional fields get left at default values even if assigned

OscarVanL, 2020-04-09:
Hi,

I'd like to report strange behaviour when I have a schema with optional fields.

Here's my schema:

```
syntax = "proto2";

message GpsPayload {
	required bool gps_fix = 1;
	required int64 epoch = 4;
	required int32 battery = 5;
	optional float latitude = 2;
	optional float longitude = 3;
	optional float speed_kph = 6;
	optional float heading = 7;
	optional float altitude = 8;
}
```

Here's my arduino code:

```
  // Buffer for storing the encoded protocol buffer message
  uint8_t buffer[128];
  GpsPayload payload = GpsPayload_init_zero;
  pb_ostream_t stream = pb_ostream_from_buffer(buffer, sizeof(buffer));
  // Set GPS data
  payload.gps_fix = gps_fix;
  payload.latitude = latitude;
  payload.longitude = longitude;
  payload.speed_kph = speed_kph;
  payload.heading = heading;
  payload.altitude = altitude;
  // Set metadata
  payload.epoch = now();
  uint16_t batt_percent;
  if (fona.getBattPercent(&batt_percent)) {
    payload.battery = batt_percent;
  }
  Serial.flush();
  
  Watchdog.reset();

  // Serialise Protocol Buffer
  bool status = pb_encode(&stream, GpsPayload_fields, &payload);
  uint8_t pb_length = stream.bytes_written;
```

With this schema, the `stream.bytes_written` is always 10 bytes whether the optional fields are set or not.

When I decode the message in Python I get this (N.B. Latitude, Longitude, Altitude should not be 0)

```
payload: \x08\x01\x20\xd3\xf5\xb9\xf4\x05\x28\x2f
GPS Fix: True
Epoch: 1586395859
Battery: 47 %
Latitude: 0.0
Longitude: 0.0
Speed km/h: 0.0
Heading: 0.0
Altitude: 0.0
```

However, when I change the schema so that optional fields are required, but make no changes to the C code it works as intended.

```
syntax = "proto2";

message GpsPayload {
	required bool gps_fix = 1;
	required int64 epoch = 4;
	required int32 battery = 5;
	required float latitude = 2;
	required float longitude = 3;
	required float speed_kph = 6;
	required float heading = 7;
	required float altitude = 8;
}
```

With this schema, the `stream.bytes_written` is significantly larger, 35 bytes.

```
When I decode the message I get this:
payload: \x08\x01\x15\xcd\x4f\x4b\x42\x1d\xcd\x08\x49\xbe\x20\x97\xf6\xb9\xf4\x05\x28\x31\x35\x00\x00\x00\x00\x3d\x00\x00\x00\x00\x45\x33\x33\x6d\x42
Decoding message:
GPS Fix: True
Epoch: 1586395927
Battery: 49 %
Latitude: 50.82793045043945
Longitude: -0.19632263481616974
Speed km/h: 0.0
Heading: 0.0
Altitude: 59.29999923706055
```

Any ideas why this is occurring?

PetteriAimonen, 2020-04-09:
In proto2 syntax and C, there is no direct way to detect when an optional field is set or not set. Nanopb generates separate fields `bool has_altitude` etc. in your struct, that you have to set to `true` if they are present.

(In C++, this is solved with setter methods. In proto3 syntax, fields are present if their value is non-zero.)

OscarVanL, 2020-04-09:
It worked. Thanks!

closed 2020-04-09 by OscarVanL