nanopb/nanopb #520: Problem with default enum value

SGrek, 2020-04-15:
I have two proto files

alignment.proto:
...
syntax = "proto2";
package alignment;
enum VerticalAlignment {
    CENTER_VERTICALLY = 0;
    BOTTOM = 1;
    TOP = 2;
} 
...

widget.proto:
...
syntax = "proto2";
package widget;

import "alignment.proto";

message Text {
    required string text = 1;
    optional alignment.VerticalAlignment vertical_alignment = 2 [default = CENTER_VERTICALLY];
...
}
...

Result execution:
D:\path_to\generator-bin>nanopb_generator.exe widget.pb
Traceback (most recent call last):
  File "<string>", line 6, in <module>
  File "__main__.py", line 128, in <module>
  File "__main__nanopb_generator__.py", line 2029, in <module>
  File "__main__nanopb_generator__.py", line 1929, in main_cli
  File "__main__nanopb_generator__.py", line 1866, in process_file
  File "__main__nanopb_generator__.py", line 1523, in generate_header
  File "__main__nanopb_generator__.py", line 1092, in fields_declaration
  File "__main__nanopb_generator__.py", line 1225, in default_value
KeyError: 'alignment_HorizontalAlignment'

If I delete `[default = CENTER_VERTICALLY]`, it's all right.
Help me understand the problem.

PetteriAimonen, 2020-04-15:
Hmm, looks like yet another problem in finding the dependent files.

Could you try this:

`nanopb_generator.exe alignment.pb widget.pb`

i.e. give both files at once to make sure it can find them.

SGrek, 2020-04-15:
The result is the same :(

SGrek, 2020-04-15:
> Could you try this:
> 
> `nanopb_generator.exe alignment.pb widget.pb`

Yes, it works!
Thank you!

closed 2020-04-15 by SGrek