nanopb/nanopb #521: ImportError: attempted relative import with no known parent package

bel2125, 2020-04-16:
Trying to build any .proto file on Windows using the batch file in release 0.4.1 fails with this "ImportError". According to some other issue reports (#470 and others), this bug was already fixed - but maybe only for Linux? Or am I just doing this wrong?

```text
D:\code\nanopb-0.4.1\generator>protoc.bat ..\tests\alltypes\alltypes.proto
```
```text
Failed to build nanopb_pb2.py: protoc --python_out=D:\code\nanopb-0.4.1\generator\proto D:\code\nanopb-0.4.1\generator\proto\nanopb.proto -I=D:\code\nanopb-0.4.1\generator\proto
Failed to build nanopb_pb2.py: protoc --python_out=D:\code\nanopb-0.4.1\generator\proto D:\code\nanopb-0.4.1\generator\proto\nanopb.proto -I=D:\code\nanopb-0.4.1\generator\proto
Traceback (most recent call last):
  File "D:\code\nanopb-0.4.1\generator\nanopb_generator.py", line 42, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\code\nanopb-0.4.1\generator\\protoc", line 4, in <module>
    from nanopb_generator import invoke_protoc
  File "D:\code\nanopb-0.4.1\generator\nanopb_generator.py", line 63, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "D:\code\nanopb-0.4.1\generator\proto\__init__.py", line 33, in <module>
    invoke_protoc(argv=cmd)
  File "D:\code\nanopb-0.4.1\generator\proto\_utils.py", line 39, in invoke_protoc
    return subprocess.check_call(argv)
  File "C:\Users\bel\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 358, in check_call
    retcode = call(*popenargs, **kwargs)
  File "C:\Users\bel\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 339, in call
    with Popen(*popenargs, **kwargs) as p:
  File "C:\Users\bel\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Users\bel\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] System cannot find the specified file

```´

PetteriAimonen, 2020-04-17:
If you are using the binary package for Windows, try it from the generator-bin folder.

For source package, try `pip install protobuf grpcio-tools` to get protoc installed.

bel2125, 2020-04-17:
Thank you - I am using the source package.
I already had `protobuf` but it seems I was missing `grpcio-tools` (maybe I misspelled it at first). However, it seems something is still missing. Building according to this documentation (https://jpa.kapsi.fi/nanopb/docs/concepts.html#compiling-proto-files-for-nanopb) does not seem to work (maybe this documentation is Linux only?):

```
C:\code\nanopb-0.4.1\tests\alltypes>python ..\..\generator\nanopb_generator.py alltypes.proto
Traceback (most recent call last):
  File "..\..\generator\nanopb_generator.py", line 2068, in <module>
    main_cli()
  File "..\..\generator\nanopb_generator.py", line 1963, in main_cli
    other_files[fdesc.name] = parse_file(fdesc.name, fdesc, options)
  File "..\..\generator\nanopb_generator.py", line 1864, in parse_file
    f = ProtoFile(fdesc, file_options)
  File "..\..\generator\nanopb_generator.py", line 1352, in __init__
    self.parse()
  File "..\..\generator\nanopb_generator.py", line 1427, in parse
    self.messages.append(Message(name, message, message_options))
  File "..\..\generator\nanopb_generator.py", line 967, in __init__
    self.load_fields(desc, message_options)
  File "..\..\generator\nanopb_generator.py", line 1007, in load_fields
    field = Field(self.name, f, field_options)
  File "..\..\generator\nanopb_generator.py", line 433, in __init__
    if field_options.HasField("type_override"):
ValueError: Protocol message NanoPBOptions has no field type_override.
```

Building using protoc.bat
```
C:\code\nanopb-0.4.1\tests\alltypes>..\..\generator\protoc.bat alltypes.proto
Missing output directives.
```

These commands don't raise an error
```
C:\code\nanopb-0.4.1\tests\alltypes>..\..\generator\protoc.bat alltypes.proto -oXY
C:\code\nanopb-0.4.1\examples\simple>..\..\generator\protoc.bat simple.proto -oXY
```
However, the XY is binary, not the `simple.pb.h` I need to compile simple.c

Probably I'm just doing this the wrong way. Is there some "getting started" documentation for Windows?

PetteriAimonen, 2020-04-17:
It looks like you are getting forwards. As for the protoc commands, use `--nanopb_out=.`. But that will probably just run into the problem you had with using `nanopb_generator.py` directly, which is usually preferred anyway.

As for the `ValueError: Protocol message NanoPBOptions has no field type_override.` error, it seems related to some new options that are new in git master, but aren't present in 0.4.1. I don't know how you could get that error if you are actually running the 0.4.1 version. Make sure that you haven't accidentally replaced `nanopb_generator.py` with another version during your testing - if you want to try a newer version, you'll need to update `nanopb.proto` also.

bel2125, 2020-04-17:
> Make sure that you haven't accidentally replaced nanopb_generator.py with another version during your testing

Yes, I tried to copy the new head version into 0.4.1 - now I switched back to the released version.

Still no success:
```
C:\code\nanopb-0.4.1\examples\simple>python ..\..\generator\nanopb_generator.py simple.c --nanopb_out=.
Usage: nanopb_generator.py [options] file.pb ...

nanopb_generator.py: error: no such option: --nanopb_out

C:\Source\nanopb-0.4.1\examples\simple>..\..\generator\protoc.bat simple.proto --nanopb_out=.
Command "protoc-gen-nanopb" not found.
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

PetteriAimonen, 2020-04-17:
Just use `python ..\..\generator\nanopb_generator.py simple.proto`

bel2125, 2020-04-17:
```
C:\code\nanopb-0.4.1\examples\simple>python ..\..\generator\nanopb_generator.py simple.c
Traceback (most recent call last):
  File "..\..\generator\nanopb_generator.py", line 2029, in <module>
    main_cli()
  File "..\..\generator\nanopb_generator.py", line 1917, in main_cli
    fdesc = descriptor.FileDescriptorSet.FromString(data).file[0]
google.protobuf.message.DecodeError: Error parsing message

```

bel2125, 2020-04-17:
Sorry: used .c instead of .proto

```
C:\Source\nanopb-0.4.1\examples\simple>python ..\..\generator\nanopb_generator.py simple.proto
C:\Users\lnr\AppData\Local\Temp\tmp33k1tp1e.pb: Permission denied
Traceback (most recent call last):
  File "..\..\generator\nanopb_generator.py", line 2029, in <module>
    main_cli()
  File "..\..\generator\nanopb_generator.py", line 1917, in main_cli
    fdesc = descriptor.FileDescriptorSet.FromString(data).file[0]
IndexError: list index (0) out of range

```

PetteriAimonen, 2020-04-17:
Ah, you're also hitting #486, which is fixed in git master but not in 0.4.1. Also, you should be feeding the .proto file, not .c file.

Sorry about all the mess, with all of Python 2 vs. 3 and Windows 7 vs. 10 vs. Linux vs. Mac and different protoc versions it's hard to test all combinations. I guess you can also try the two-step variant of `....protoc -osimple.pb simple.proto` `nanopb_generator.py simple.pb`.

bel2125, 2020-04-17:
Thanks a lot - it seems to work using the two step variant.

```
C:\code\nanopb-0.4.1\examples\simple>..\..\generator\protoc.bat -osimple.pb simple.proto

C:\code\nanopb-0.4.1\examples\simple>python ..\..\generator\nanopb_generator.py simple.pb
Writing to simple.pb.h and simple.pb.c

C:\code\nanopb-0.4.1\examples\simple>type simple.pb.h
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.1 */

#ifndef PB_SIMPLE_PB_H_INCLUDED
#define PB_SIMPLE_PB_H_INCLUDED
#include <pb.h>
...
```

And, to get startet,  once `python -m pip install protobuf grpcio-tools`

PetteriAimonen, 2020-04-17:
Great :)

I should work on getting 0.4.2 released with the fixes for Windows. I guess the update to the online readme was too early, the instructions in the README.md that comes with 0.4.1 work on Windows also but the online instructions only work on Linux or git master.

closed 2020-06-23 by PetteriAimonen

wtford, 2020-09-07:
I am running into the same error.  Mac.  Nanopb from source checked out at 0.4.2.

`~/git/nanopb/examples/simple$ ../../generator/nanopb_generator.py simple.proto
Traceback (most recent call last):
  File "../../generator/nanopb_generator.py", line 50, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "../../generator/nanopb_generator.py", line 71, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/Users/walker/git/nanopb/generator/proto/nanopb_pb2.py", line 9, in <module>
    from google.protobuf import symbol_database as _symbol_database
ImportError: cannot import name 'symbol_database' from 'google.protobuf' (/usr/local/lib/python3.8/site-packages/google/protobuf/__init__.py)`

Git log
`~/git/nanopb/examples/simple$ git log -1
commit df0e92f474f9cca704fe2b31483f0b4d1b1715a4 (HEAD, tag: nanopb-0.4.2, tag: 0.4.2)
Author: Petteri Aimonen <jpa@git.mail.kapsi.fi>
Date:   Tue Jun 23 15:03:40 2020 +0300

    Publishing nanopb-0.4.2`

Pip3 state:
`~/git/nanopb/examples/simple$ pip3 list
Package          Version
---------------- -------
beautifulsoup4   4.9.1
google           3.0.0
grpcio           1.31.0
grpcio-tools     1.31.0
pip              20.1.1
protobuf         3.13.0
PyQt5            5.15.0
PyQt5-sip        4.19.24
python3-protobuf 2.5.0
setuptools       49.2.0
six              1.15.0
soupsieve        2.0.1
wheel            0.34.2`

Protoc version:
`~/git/nanopb/examples/simple$ protoc --version
libprotoc 3.13.0`

wtford, 2020-09-07:
Using the binary release works fine:
```
~/Downloads/nanopb-0.4.2-macosx-x86/examples/simple$ ../../generator-bin/protoc --nanopb_out=. simple.proto
~/Downloads/nanopb-0.4.2-macosx-x86/examples/simple$ ls
Makefile     README.txt   simple.c     simple.pb    simple.pb.c  simple.pb.h  simple.proto

wtford, 2020-09-07:
I was able to fix the above problem, which was caused by the older `python3-protobuf` python module that was installed on my system and was taking precedence over the other protobuf library.

wtford, 2020-09-07:
But I am having the following issue, when trying to compile the proto files using the generator:
```
~/git/nanopb/examples/network_server$ python3 ../../generator/nanopb_generator.py fileproto.proto 
Writing to fileproto.pb.h and fileproto.pb.c
Traceback (most recent call last):
  File "../../generator/nanopb_generator.py", line 2122, in <module>
    main_cli()
  File "../../generator/nanopb_generator.py", line 2034, in main_cli
    os.makedirs(os.path.dirname(path), exist_ok=True)
  File "/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/os.py", line 223, in makedirs
    mkdir(name, mode)
FileNotFoundError: [Errno 2] No such file or directory: ''
```

wtford, 2020-09-07:
`make` correctly generates the pb files, although it deletes the "fileproto.pb.c" because make thinks it is build artifact:

```
~/git/nanopb/examples/network_server$ make
../../generator/protoc  --nanopb_out=. fileproto.proto
cc -ansi -Wall -Werror -g -O0 -I../.. -o server server.c common.c fileproto.pb.c ../../pb_encode.c ../../pb_decode.c ../../pb_common.c
cc -ansi -Wall -Werror -g -O0 -I../.. -o client client.c common.c fileproto.pb.c ../../pb_encode.c ../../pb_decode.c ../../pb_common.c
rm fileproto.pb.c
```

wtford, 2020-09-07:
The problem is corrected if you change line 2023 of nanopb_generator.py from:
```
        base_dir = options.output_dir or ''
```
to:
```
        base_dir = options.output_dir or '.'
```

PetteriAimonen, 2020-09-07:
@wtford Yeah, that issue is #553.

AgentCoop, 2020-09-10:
As a solution:
PYTHONPATH=../../lib/nanopb/generator protoc --nanopb_out=. tmp/message.prot