nanopb/nanopb #522: Compile Errors when using UENUM on PlatformIO

richardjlyon, 2020-04-19:
I am building a program for an embedded system using PlatformIO. When compiling, I hit errors using an enum in a message.

```yaml
[env:feather32u4]
platform = atmelavr
board = feather32u4
framework = arduino
```

Example proto file:

```proto
syntax = "proto3";

enum ReadingType {
   UNKNOWN = 0;
   TEMP = 1;
   HUMIDITY = 2;
   PRESSURE = 3;
   CO2 = 4;
   LIGHT = 5;
   POWER = 6;
}

message Reading {
   ReadingType readingType = 1;
   float floatVal = 2;
}

message Lora {
   float rssi = 1;
   float frequency_error = 2;
   float snr = 3;
}

message Meta {
   int32 device_id = 1;
   string device_type = 2;
   float voltage = 3;
   Lora lora =  4;
}

message Packet {
   int32 packet_id = 1;
   Meta meta = 2;
   repeated Reading data = 3;
}  
```

The ReadingType is encoded as a `UENUM`

```c
/* Struct field encoding specification for nanopb */
#define Reading_FIELDLIST(X, a)                   \
    X(a, STATIC, SINGULAR, UENUM, readingType, 1) \
    X(a, STATIC, SINGULAR, FLOAT, floatVal, 2)
#define Reading_CALLBACK NULL
#define Reading_DEFAULT NULL
```

This fails when compiling:

```
Compiling .pio/build/feather32u4/lib6e7/Protobuf/packet.pb.c.o
In file included from lib/Protobuf/packet.pb.h:6:0,
                 from lib/Protobuf/packet.pb.c:4:
.pio/libdeps/feather32u4/Nanopb/pb.h:734:44: error: pasting "PB_FIELDINFO_WIDTH_" and "(" does not give a valid preprocessing token
 #define PB_FIELDINFO_WIDTH_SINGULAR(ltype) PB_FIELDINFO_WIDTH_ ## ltype
                                            ^
.pio/libdeps/feather32u4/Nanopb/pb.h:730:49: note: in expansion of macro 'PB_FIELDINFO_WIDTH_SINGULAR'
[... 243 lines cut ...]
lib/Protobuf/packet.pb.c:9:1: note: in expansion of macro 'PB_BIND'
 PB_BIND(Reading, Reading, AUTO)
 ^
.pio/libdeps/feather32u4/Nanopb/pb.h:526:82: error: pasting "PB_LTYPE_MAP_" and "(" does not give a valid preprocessing token
     PB_FIELDINFO_ASSERT_ ## width(tag, PB_ATYPE_ ## atype | PB_HTYPE_ ## htype | PB_LTYPE_MAP_ ## ltype, \
                                                                                  ^
.pio/libdeps/feather32u4/Nanopb/pb.h:523:5: note: in expansion of macro 'PB_GEN_FIELD_INFO_ASSERT'
     PB_GEN_FIELD_INFO_ASSERT(width, structname, atype, htype, ltype, fieldname, tag)
     ^
.pio/libdeps/feather32u4/Nanopb/pb.h:520:5: note: in expansion of macro 'PB_GEN_FIELD_INFO_ASSERT_AUTO2'
     PB_GEN_FIELD_INFO_ASSERT_AUTO2(PB_FIELDINFO_WIDTH_AUTO(atype, htype, ltype), structname, atype, htype, ltype, fieldname, tag)
     ^
lib/Protobuf/packet.pb.h:113:5: note: in expansion of macro 'PB_GEN_FIELD_INFO_ASSERT_AUTO'
     X(a, STATIC, SINGULAR, UENUM, readingType, 1) \
     ^
.pio/libdeps/feather32u4/Nanopb/pb.h:478:5: note: in expansion of macro 'Reading_FIELDLIST'
     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
     ^
lib/Protobuf/packet.pb.c:9:1: note: in expansion of macro 'PB_BIND'
 PB_BIND(Reading, Reading, AUTO)
 ^
*** [.pio/build/feather32u4/lib6e7/Protobuf/packet.pb.c.o] Error 1
```

Interestingly, changing the generated code to use ENUM instead compiles correctly.

PetteriAimonen, 2020-04-19:
Hmm, the error does not occur with avr-gcc directly.

Are you somehow able to see the full command line it uses to compile the file?

PetteriAimonen, 2020-04-19:
Ah, now I know what is causing it: atmega32u4 has a register called `UENUM`, controlling USB enumeration.

I'll have to think about how to best fix this.

richardjlyon, 2020-04-19:
Thanks for the quick reply - I’m up and running with signed ENUMs for now. Best wishes

[Richard Lyon - Chat @ Spike](https://spikenow.com/r/a/?ref=spike-organic-signature&_ts=h1dg0)	[h1dg0]

richardjlyon, 2020-04-22:
Using signed enums was adding about 50 bytes to the packet, so after some experimentation I found a workaround. Not sure what the implications are, but I'm not using USB.

```diff
diff --git a/file.pb.h b/file.pb.h
index 68cb7a5..e49209c 100644
--- a/file.pb.h
+++ b/file.pb.h
@@ -5,6 +5,10 @@
 #define PB_PACKET_PB_H_INCLUDED
 #include <pb.h>
 
+#ifdef UENUM
+#undef UENUM
+#endif
+
 #if PB_PROTO_HEADER_VERSION != 40   
 #error Regenerate this file with the current version of nanopb generator.
 #endif
```

closed 2020-06-23 by PetteriAimonen