nanopb/nanopb #527: CMake NANOPB_OUT relation to FIL_PATH_REL

BrianPugh, 2020-04-28:
Whats the current deal with the output directory `NANOPB_OUT` in CMake with the `RELPATH` option? In the current master branch of `nanopb`, my `.pb.c` and `.pb.h` files are being outputted to the root of `CMAKE_CURRENT_BINARY_DIR` instead of their relative paths.

For example, if my proto file is located in `my_project/my_proto_files/foo.proto`, I'd expect the output files to be located at `${CMAKE_CURRENT_BINARY_DIR}/my_proto_files/foo.pb.*`. However, I find them at `${CMAKE_CURRENT_BINARY_DIR}/foo.pb.*`

The issue seems to be the specific output directory to `protoc`:
https://github.com/nanopb/nanopb/blob/beab2e8ca9ac9a55595d210b7314de52572c987c/extra/FindNanopb.cmake#L253

```
# based on the version of protoc it might be necessary to add "/${FIL_PATH_REL}" currently dealt with in #516
```

The comment above it:
https://github.com/nanopb/nanopb/blob/beab2e8ca9ac9a55595d210b7314de52572c987c/extra/FindNanopb.cmake#L252
 fixes my issues. To be clear, meaning 
```
set(NANPB_OUT "${CMAKE_CURRENT_BINARY_DIR}/${FIL_PATH_REL}")
```
Is this a wide spread issue, or something specific to my setup? I see that `${FIL_PATH_REL}` has been added and reverted in the commit history; so whats the deal with it?

**My current setup:**
OS: Ubuntu 18.04
protoc: 3.11.4 (I have also observed the same issue/solution on 3.0.0)
cmake: 3.10.2

PetteriAimonen, 2020-04-28:
Honestly I'm just as confused about the CMake stuff as you. I'm not aware of behavior differences here based on protoc version, but they might exist.

@efisher Any ideas?

efisher, 2020-04-29:
I'm struggling to understand it. My change that broke others made sense to me. I am currently using a fork of nanopb with my patch applied.

Could it be related to this?
https://github.com/protocolbuffers/protobuf/issues/3044

PetteriAimonen, 2020-04-30:
Oh, that could very well be related! @BrianPugh Can you find out the actual arguments CMake is giving to protoc in your case?

BrianPugh, 2020-04-30:
So if i revert my mod (so stock nanopb) i get the following (and incorrectly located produced pbc/pbh files). I added line breaks for readability, but this is all one command

```
cd /home/bpugh/esp/jolt_wallet/build/esp-idf/jolt_os_test 
&& /usr/local/bin/protoc 
-I/home/bpugh/esp/jolt_wallet/build/nanopb/generator
-I/home/bpugh/esp/jolt_wallet/build/nanopb/generator/proto 
-I/home/bpugh/esp/jolt_wallet/build/esp-idf/jolt_os_test 
-I/home/bpugh/esp/jolt_wallet/jolt_os/test/nanopb 
-I/home/bpugh/esp/jolt_wallet/jolt_os/test 
--plugin=protoc-gen-nanopb=/home/bpugh/esp/jolt_wallet/build/nanopb/generator/protoc-gen-nanopb 
--nanopb_opt= 
--nanopb_out=/home/bpugh/esp/jolt_wallet/build/esp-idf/jolt_os_test /home/bpugh/esp/jolt_wallet/jolt_os/test/nanopb/simple.proto
```

where

```
-I/home/bpugh/esp/jolt_wallet/build/esp-idf/jolt_os_test 
```

is the root of the build for this specific component. I wanted the produced `simple.pb.*`  files to be located at 

```
/home/bpugh/esp/jolt_wallet/build/esp-idf/jolt_os_test/nanopb/simple.pb.*
```

I'll investigate why `-I/home/bpugh/esp/jolt_wallet/jolt_os/test/nanopb` is being passed in, because i think thats erroneous and could be causing this.

PetteriAimonen, 2020-04-30:
Looks like it is probably related to option `NANOPB_GENERATE_CPP_APPEND_PATH`. Setting it to false should disable adding the path of each file into include.

The protoc include path handling seems to be an issue that divides opinions. Some consider that file's own include path should be in the path automatically, others don't. Then protoc goes generating the relative path based on that, which causes further confusion.

Just recently in #499 similar logic was suggested to be added to nanopb_generator.py, but I opted not to because I wanted to keep the behavior close to protoc.

The logic has been in the FindNanopb.cmake since pull request #273.

BrianPugh, 2020-05-01:
@PetteriAimonen that solves my issues! Adding

```
set(NANOPB_GENERATE_CPP_APPEND_PATH FALSE)
```

to my calling CMakeLists.txt prevents the full proto directories from being passed to `protoc`, resulting in my `.pb.*` files being built to my desired build directories. Thanks!

I'd say we should update the documentation in regards to this setting to be a bit more explicit.

PetteriAimonen, 2020-05-01:
I updated the comment in FindNanopb.cmake now based on my understanding of the issue.

BrianPugh, 2020-05-01:
I agree with the updated documentation, thank you so much!

closed 2020-05-01 by BrianPugh