nanopb/nanopb #528: Specify maximum total encoded message size

sureshjoshi, 2020-04-28:
**What should the feature do?**

I was wondering if there was any facility to specify (in the proto files, or options) the total desired encoded message size. As in, not just the sum of the maximum components of the message, but to reverse that to limit the internal capacities of a string or byte array. 

**In what situation would the feature be useful?**

In my case, for instance, I have flash blocks with fixed 4096 bytes, so what I'm currently doing is encoding various protobuffers and manually writing them into my flash. But, since a lot of PBs can be variably sized (which is a benefit), I end up needing to add a framing byte as well so I can find them later.

Instead, I was wondering if there was some way to do something like this:

In flash.proto:
```
message FlashBlock {
    message Header {
        uint32 foo = 1;
        fixed32 checksum = 2;
        ...
    }
    Header header = 1;
    bytes data = 2;
}
```

In flash.options (something like what I would envision):
```
FlashBlock            fixed_length:true max_size:4096
FlashBlock.data   fixed_length:true max_size:*
```

Where `*` represents taking up all available space defined in the message, other than what is already occupied by existing fields.

As of today, if I set `FlashBlock.data max_size:4096`, then I get

```
    ...
    pb_byte_t data[4096];
    ...

#define FlashBlock_size                        4114
#define FlashBlock_Header_size               13
```

Alternatively, if I solely have a field for `data`, then I get:

```
    ...
    pb_byte_t data[4096];
    ...

#define FlashBlock_size                        4099
```

I know this is easily reconciled in code, or I can manually specify a different `max_size` but, if I use the former method, and then someone comes along and updates Header with another field, then doesn't update `max_size` - all hell breaks loose.

PetteriAimonen, 2020-04-28:
I've had the same need myself, though with need to pad up to next 512 bytes to fill up complete USB packets. It might be possible to add something like `pb_pad_message(&stream, desired_size);`, but I'm not sure if the need is general enough to be included in main nanopb code.

So far I've used code like this in my own application:

    message USBResponse {
         .....
         bytes padding = 31; // Ignore this field, it is used to pad message size
    }

.

            uint32_t len = stream->bytes_written;
            if (len % 512 != 0)
            {
                // Pad the message to 512 bytes. Sending partial USB packets can cause the host
                // to delay the next poll, causing lower throughput. We could also copy the beginning
                // of the next packet instead, but that would cause unaligned accesses. In most
                // cases it is easier to just insert a few dummy bytes.
                size_t padding = 512 - (len % 512);

                if (padding == 2)
                {
                    pb_encode_tag(&stream, PB_WT_STRING, USBResponse_padding_tag);
                    pb_encode_varint(&stream, 0);
                    len += padding;
                }
                else if (padding >= 3)
                {
                    pb_encode_tag(&stream, PB_WT_STRING, USBResponse_padding_tag);
                    size_t bytes_len = padding - 3;
                    // Encode varint length so that it always takes up two bytes.
                    uint8_t varint[2] = {0x80 | (bytes_len & 0x7F), (bytes_len >> 7)};
                    pb_write(&stream, varint, 2);
                    // We don't actually write the padding bytes here, let it take garbage from RAM.
                    len += padding;
                }
            }

sureshjoshi, 2020-04-28:
Thanks for the quick reply @PetteriAimonen 

In my case, it's not padding, it's the underlying data that goes into the `data` field, and I don't want to exceed the flash block size, otherwise 💥 

In thinking about it, even if there were some extra defines in the .pb.h file, that could work, or a way to reference max sizes from the .options file.

For instance, if there was the ability to specify the max_size of the message, then we could have:

```
#define FlashBlock_size                        4096 // Specified by the .options, failure to generate otherwise
#define FlashBlock_Header_size          13
#define FlashBlock_data_size               FlashBlock_size - FlashBlock_Header_size

    ...
    pb_byte_t data[FlashBlock_data_max_size];
    ...
```

On a similar note, is there any reason why the .options max_size isn't reflected in a #define inside the .pb.h file, and is hardcoded? I just noticed that for the first time earlier today.

PetteriAimonen, 2020-04-28:
Ah, ok. Seems like a quite specific need, but then again it is generator-only so doesn't cost code size.

You can get the `.options` max size with `sizeof(mymsg.field)`.

sureshjoshi, 2020-04-28:
Ah, right, sorry, I meant inside of the .options file to get the max size of a field - and use it as the max_size of another field.

I'll keep investigating this.

sureshjoshi, 2020-04-28:
In case anyone else runs into this issue: 

In the meantime, I've just been using a compile-time assert to ensure I don't accidentally re-generate and compile with incorrect message sizes.

PetteriAimonen, 2020-04-28:
Yeah, the .options syntax is not really up to much scripting as it is just a protobuf message on each line.

One option could be to have the generator put `pb_byte_t field[MYMESSAGE_FIELD_SIZE]` in the header, and `MYMESSAGE_FIELD_SIZE` could be defined to anything needed. But that would probably run into an issue with the order of definition, as you'd need to define the size before including the header and you don't know the size beforehand.

sureshjoshi, 2020-05-05:
I haven't quite solved the issue as I've listed it, but a short term workaround that I mentioned earlier:

```c
static_assert(FlashBlock_size <= MY_MAX_SIZE, "FlaskBlock PB is too large");
```

PetteriAimonen, 2024-09-19:
I'm closing this off as I don't see a reasonable way to implement this.

closed 2024-09-19 by PetteriAimonen