nanopb/nanopb #529: proto_path option does not work on Windows

yogeshkm20, 2020-05-01:
This is similar to proto_path option does not work on Windows #362. 
C:\Users\ykmh\Downloads\nanopb-0.4.1-windows-x86>.\generator-bin\protoc.exe --nanopb_out=.\examples\simple --python_out=.\examples\simple --proto_path=.\examples\simple .\examples\simple\*.proto
.\examples\simple\*.proto: No such file or directory

C:\Users\ykmh\Downloads\nanopb-0.4.1-windows-x86>dir .\examples\simple
 Volume in drive C is Windows
 Volume Serial Number is F656-CC0B

 Directory of C:\Users\ykmh\Downloads\nanopb-0.4.1-windows-x86\examples\simple

02/02/2020  03:52 PM    <DIR>          .
02/02/2020  03:52 PM    <DIR>          ..
02/02/2020  03:52 PM               742 Makefile
02/02/2020  03:52 PM               877 README.txt
02/02/2020  03:52 PM             2,289 simple.c
02/02/2020  03:52 PM               167 simple.proto
               4 File(s)          4,075 bytes
               2 Dir(s)  76,132,073,472 bytes free

PetteriAimonen, 2020-05-01:
Hmm, what if instead of `*.proto`, you give `simple.proto`?

Windows does not automatically expand wildcards if you use the legacy command prompt, though I think it should expand them if you use powershell.

yogeshkm20, 2020-05-04:
That works. I have to list out each file on the command line.

yogeshkm20, 2020-05-04:
I am able to list out the files in 0.3.9.2 which I was using earlier

closed 2020-06-21 by PetteriAimonen