nanopb/nanopb #530: Default values of +-Infinity and NaN not supported

jrullman, 2020-05-06:
A floating-point field marked with [default = inf] will result in generated code with default value 'inf', which doesn't compile. According to https://developers.google.com/protocol-buffers/docs/reference/proto2-spec#floating-point-literals, 'inf' and 'nan' are allowed values. The generator script should perform the following replacements on default values for floating point fields:

inf -> INFINITY
-inf -> -INFINITY
nan -> NAN
-nan -> -NAN

INFINITY and NAN are defined in math.h.

jrullman, 2020-05-06:
(INFINITY has to be cast to float in the single-precision case)

PetteriAimonen, 2020-05-06:
Sounds reasonable.

Currently `math.h` isn't being included, so the include should be added by the generator only if these default values are found in the file.

closed 2020-06-23 by PetteriAimonen