nanopb/nanopb #532: Decoding doesn't set default enum values correctly in proto2

jrullman, 2020-05-13:
According to https://developers.google.com/protocol-buffers/docs/proto#optional:
"For enums, the default value is the first value listed in the enum's type definition."

For a proto2 message containing an enum field, the "_init_default" struct initializer follows the above rule, but the decoder does not. No extern const "_default" value for the enum is generated, and the decoder will set the enum field to zero when the bytes don't contain that field.

PetteriAimonen, 2020-05-13:
Ah, this was missed when #295 was fixed.

A possible workaround is specifying the default value explicitly.

jrullman, 2020-05-13:
The most efficient fix would be to have a global for the default value of any enum (when it's not 0), rather than having to add a global for any field of enum type.

PetteriAimonen, 2020-05-13:
The way 0.4.x handles default values it would just take 1-2 bytes in the protobuf encoded representation of the defaults.

closed 2020-06-23 by PetteriAimonen