nanopb/nanopb #543: Self-tests failure on some architectures

gcsideal, 2020-06-21:
Hi,
nanopb packaged for Debian and builds on all architectures it supports. But [self-tests are failing](https://buildd.debian.org/status/package.php?p=nanopb&suite=sid) on some architectures as:
run_test(["build/float_double_conversion/float_double_conversion.output"], ["build/float_double_conversion/float_double_conversion"])
Command line: ['build/float_double_conversion/float_double_conversion']
FAILED:[22;39m build/float_double_conversion/float_double_conversion.c:62 memcmp(&msg.value, &expected_float, sizeof(float)) == 0
FAILED:[22;39m build/float_double_conversion/float_double_conversion.c:76 memcmp(&msg.value, &expected_double, sizeof(double)) == 0
FAILED:[22;39m build/float_double_conversion/float_double_conversion.c:62 memcmp(&msg.value, &expected_float, sizeof(float)) == 0
FAILED:[22;39m build/float_double_conversion/float_double_conversion.c:76 memcmp(&msg.value, &expected_double, sizeof(double)) == 0
[FAIL][0m   Program build/float_double_conversion/float_double_conversion returned 1

I don't know if it's a coincidence, but the failing architectures are big-endian and the comparison may fail due to the byte order.

PetteriAimonen, 2020-06-21:
Hmm, it could be endianess issue in the PB_CONVERT_DOUBLE_FLOAT 
functionality. I *think* it should work for big-endian also, but
it would certainly break for mixed-endian architectures where FPU 
endianess would differ from integer endianess.

It's not an option that would be used on most architectures, it is 
mostly for AVR which has no 64-bit double type. Currently the test runs 
on all architectures but it certainly doesn't have to.

I'll have to look up if there would be some C compiler define or similar 
I could use to detect this.

Also, thanks for the packaging. I took a brief look and they look 
reasonable :)

PetteriAimonen, 2020-06-21:
Turns out it was a portability bug in the PB_CONVERT_DOUBLE_FLOAT 
implementation, which failed to handle NaN representations with bottom
bits set.

The test cases succeed for mips and mipsel for me now.

gcsideal, 2020-06-21:
> Turns out it was a portability bug in the PB_CONVERT_DOUBLE_FLOAT implementation, which failed to handle NaN representations with bottom bits set. The test cases succeed for mips and mipsel for me now.

Just tested current 'master' on one of our mipsel machines. Unable to confirm, fails the same way.
Double checked that the two new commits are in place and triple compiled. How can I help you?

PetteriAimonen, 2020-06-22:
Hmm, here is what you can do:

1. Run `tests/build/float_double_conversion/float_double_conversion` to see full output instead of just stderr. Check that the failing test case is actually the `nan` testcase as it was for me.

2. Apply this patch on `tests/float_double_conversion/float_double_conversion.c` to get some more debug prints for me:

<pre>
--- float_double_conversion.c	2020-06-21 20:13:34.258665221 +0300
+++ /home/petteri/snippets/2020-06-21T20-13-..-..-..-float_double_conversion-float_double_conversion.c	2020-06-21 19:55:27.237103359 +0300
@@ -34,7 +34,7 @@
     uint8_t buf[16];
     size_t msglen;
     int status = 0;
-    int i;
+    int i, j;
     for (i = 0; i < TESTVALUES_COUNT; i++)
     {
         double orig_double = testvalues[i];
@@ -60,6 +60,9 @@
             FloatMsg msg = { 0.0f };
             TEST(pb_decode(&stream, &FloatMsg_msg, &msg));
             TEST(memcmp(&msg.value, &expected_float, sizeof(float)) == 0);
+            for (j = 0; j < msglen; j++) printf("%02x ", buf[j]);
+            printf("\n");
+            printf("%016llx %08x %08x\n", *(uint64_t*)&orig_double, *(uint32_t*)&msg.value, *(uint32_t*)&expected_float);
 
             /* Re-encode */
             ostream = pb_ostream_from_buffer(buf, sizeof(buf));
</pre>

gcsideal, 2020-06-23:
>     1. Run `tests/build/float_double_conversion/float_double_conversion` to see full output instead of just stderr. Check that the failing test case is actually the `nan` testcase as it was for me.

It's nan failing for me, even with the 0.4.2 release.

> 
>     2. Apply this patch on `tests/float_double_conversion/float_double_conversion.c` to get some more debug prints for me:

Patch applied and output:
---- Testcase: nan ----
OK: pb_encode(&stream, &DoubleMsg_msg, &msg)
OK: msglen == 9
OK: pb_decode(&stream, &FloatMsg_msg, &msg)
OK: memcmp(&msg.value, &expected_float, sizeof(float)) == 0
09 ff ff ff ff ff ff f7 7f
7ff7ffffffffffff 7fbfffff 7fbfffff
OK: pb_encode(&ostream, &FloatMsg_msg, &msg)
OK: msglen == 9
OK: pb_decode(&stream, &DoubleMsg_msg, &msg)
FAILED: build/float_double_conversion/float_double_conversion.c:79 memcmp(&msg.value, &expected_double, sizeof(double)) == 0

---- Testcase: nan ----
OK: pb_encode(&stream, &DoubleMsg_msg, &msg)
OK: msglen == 9
OK: pb_decode(&stream, &FloatMsg_msg, &msg)
OK: memcmp(&msg.value, &expected_float, sizeof(float)) == 0
09 ff ff ff ff ff ff f7 7f
7ff7ffffffffffff 7fbfffff 7fbfffff
OK: pb_encode(&ostream, &FloatMsg_msg, &msg)
OK: msglen == 9
OK: pb_decode(&stream, &DoubleMsg_msg, &msg)
FAILED: build/float_double_conversion/float_double_conversion.c:79 memcmp(&msg.value, &expected_double, sizeof(double)) == 0

It's a four core mipsel:
system type	        : cavium,rhino_utm8 (CN7030p1.2-1500-AAP)
cpu model	        : Cavium Octeon III V0.2  FPU V0.0

PetteriAimonen, 2020-06-23:
Ah, looks I've fixed the double->float direction, but not float->double direction.
Apparently the floating point implementation my qemu-based test ends up using fills the bottom bits with zero like my code does, but the hardware FPU fills them with ones.

I've now updated the test so that it doesn't care too much about the nan representation, just that it is nan.

gcsideal, 2020-06-24:
> Ah, looks I've fixed the double->float direction, but not float->double direction.
> Apparently the floating point implementation my qemu-based test ends up using fills the bottom bits with zero like [...]

Thanks, nan self-tests are working now on all architectures. Only SPARC64 fails self-testing now. Reason is ./build/alltypes_pointer/decode_alltypes_pointer segfaults before decoding anything (no output). Just started gdb which reveals:
Program received signal SIGSEGV, Segmentation fault.
pb_field_iter_begin_extension (iter=0x7feffffe778, extension=0xffffff01b2010e00) at ./pb_common.c:168
168	    const pb_msgdesc_t *msg = (const pb_msgdesc_t*)extension->type->arg;

But I let it go.

PetteriAimonen, 2020-06-25:
Thanks for the info. I may try with qemu-sparc64 at some point.