nanopb/nanopb #548: Bug when encoding optional fields (proto2 only)

logsol, 2020-06-22:
I encountered this issue, that the encoding fails for fields which are set to optional in the proto2 syntax.

I followed the code down to where the issue appears and it looks like it is always returning true with the code comment "Missing optional field".

Once the `return true;` is commented out it seems to work. 

```
/* Encode a single field of any callback, pointer or static type. */
static bool checkreturn encode_field(pb_ostream_t *stream, pb_field_iter_t *field)
{
    /* Check field presence */
    if (PB_HTYPE(field->type) == PB_HTYPE_ONEOF)
    {
        if (*(const pb_size_t*)field->pSize != field->tag)
        {
            /* Different type oneof field */
            return true;
        }
    }
    else if (PB_HTYPE(field->type) == PB_HTYPE_OPTIONAL)
    {
        if (field->pSize)
        {
            if (safe_read_bool(field->pSize) == false)
            {
                /* Missing optional field */
                return true;    // <================== here
            }
        }
        else if (PB_ATYPE(field->type) == PB_ATYPE_STATIC)
        {
            /* Proto3 singular field */
            if (pb_check_proto3_default_value(field))
                return true;
        }
    }
```

Could you give some advice as to what is going on or if I am using the API wrong?

I am not getting any errors, just an empty buffer. This is my encoding code:
```
    bool status;
    ProtoNotification notification = ProtoNotification_init_zero;
    
    /* Create a stream that will write to our buffer. */
    pb_ostream_t stream = pb_ostream_from_buffer(buffer, *length);
    
    /* Fill in the data */
    char* app_name = "WhatsApp";
    char* title = "Message for you, Sir!";
    char* subtitle = "Juliet";
    char* message = "O Romeo, Romeo, wherefore art thou Romeo?";

    strcpy(notification.app_name, app_name);
    strcpy(notification.title, title);
    strcpy(notification.subtitle, subtitle);
    strcpy(notification.message, message);
    
    /* Now we are ready to encode the notification! */
    status = pb_encode(&stream, ProtoNotification_fields, &notification);
    
    /* Then just check for any errors.. */
    if (!status)
    {
        printf("Encoding failed: %s\n", PB_GET_ERROR(&stream));
        return 1;
    }
```

PetteriAimonen, 2020-06-22:
You have to set `notification.has_app_name = true;` and similar.

logsol, 2020-06-23:
Thank you that works!

closed 2020-06-23 by logsol

chen0411, 2021-08-13:
I have the same problem in proto3, how to solve it ?

chen0411, 2021-08-13:
> You have to set `notification.has_app_name = true;` and similar.

I have the same problem in proto3, how to solve it ?