nanopb/nanopb #551: pb_decode clears callbacks in pb_callback_t fields within oneof's

jwhong, 2020-06-26:
Hi, 

*Problem:*
When decoding a message which has callback dependent fields within a oneof, `pb_decode_static_field` will re-initialize the selected oneof and remove any attached callbacks, causing those fields to be skipped in the decoding.

*Solution:*
I think this issue can be solved strictly by removal of the following code: pb_decode.h lines [468, 474]:

```c
if (PB_LTYPE(type) == PB_LTYPE_SUBMESSAGE)
{
  /* We memset to zero so that any callbacks are set to NULL.
  * Then set any default values. */
  memset(iter->pData, 0, iter->pos->data_size);
  pb_message_set_to_defaults((const pb_field_t*)iter->pos->ptr, iter->pData);
}```

PetteriAimonen, 2020-06-26:
Nope, see #175. Because all the subfields occupy the same memory, the callbacks have to be cleared to avoid crashing.
In 0.4.x, there are better ways to handle this.

jwhong, 2020-06-26:
It seemed to me that since you know the `which` tag of the union, you know which member of the union will actually be used in the decoding of the submessage, you could just set the callback of the active union element and leave the others alone. There's probably a use case I'm missing...

Thank you, I will look into the `no_unions` and `submsg_callback` options.  Appreciate your work!

closed 2020-06-26 by jwhong

PetteriAimonen, 2020-06-27:
The issue is that within an union, all the members occupy the same 
memory. So if you store a callback in different submessages inside
oneof, they could overwrite each other.

In simple cases it would be possible for exactly one oneof member
to contain callbacks, but even that would break if the protobuf
message would contain a different oneof member before the 
callback-containing one. That's not very common, but it is valid 
protobuf.

SoylentGraham, 2022-05-31:
And just to complete the solution; in the new callback for the `union`/`oneof`, one can assign typical callbacks that are normally null'd
```
bool DecodeSegmentField(pb_istream_t *stream, const pb_field_t *field, void **arg)
{
	auto& Segment = *reinterpret_cast<Segment_t*>(arg[0]);
	
	if ( field->tag == myproto_Segment_external_tag )
	{
		auto& Message = *reinterpret_cast<myproto_Segment_external*>( field->message );
		Message.textures.external.url.funcs.decode = DecodeToStdString;
		Message.textures.external.url.arg = &Segment.mExternalUrl;
		Message.textures.external.content_type.funcs.decode = DecodeToStdString;
		Message.textures.external.content_type.arg = &Segment.mExternalContentType;
		return true;
	}
	else
	{
		return false;
	}
}
```