nanopb/nanopb #552: Option to skip identifier package prefix

benma, 2020-06-27:
Hi

When there is a `package foo.bar;` in the proto files, all the C structs, defines, enums etc. have names prefixed with `foo_bar_`.

There is the `package` option to replace that with something else, but it seems it cannot be empty to strip the prefix.

Is there a way to strip the package prefix in names everywhere? In my project I know that there will be no name clash with other protobuf files, so I don't want to deal with very long names.

benma, 2020-06-27:
Digging through the code I found this undocumented option:

`mangle_names:M_STRIP_PACKAGE`

Please consider documenting it.

closed 2020-06-27 by benma

PetteriAimonen, 2020-06-27:
It's documented in the lazy way: https://jpa.kapsi.fi/nanopb/docs/reference.html#proto-file-options "Here is a list of the most common options, but see the file for a full list"