nanopb/nanopb #557: Field name `a` not supported

egmontkob, 2020-07-06:
Hi,

I'm new to this project, and of course my first attempt failed with such cryptic messages while compiling `.pb.c`:

```
In file included from simple.pb.h:6,
                 from simple.pb.c:4:
simple.pb.c:9:1: error: ‘SimpleMessage’ {aka ‘struct _SimpleMessage’} has no member named ‘SimpleMessage’
    9 | PB_BIND(SimpleMessage, SimpleMessage, AUTO)
      | ^~~~~~~
/usr/include/pb.h:462:9: note: in expansion of macro ‘SimpleMessage_FIELDLIST’
  462 |         msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ ## width, structname) \
      |         ^~~~~~~
simple.pb.c:9:1: error: ‘SimpleMessage’ {aka ‘struct _SimpleMessage’} has no member named ‘SimpleMessage’
    9 | PB_BIND(SimpleMessage, SimpleMessage, AUTO)
      | ^~~~~~~
/usr/include/pb.h:462:9: note: in expansion of macro ‘SimpleMessage_FIELDLIST’
  462 |         msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ ## width, structname) \
      |         ^~~~~~~
In file included from simple.pb.h:6,
                 from simple.pb.c:4:
simple.pb.c:9:1: error: ‘SimpleMessage’ {aka ‘struct _SimpleMessage’} has no member named ‘SimpleMessage’
    9 | PB_BIND(SimpleMessage, SimpleMessage, AUTO)
      | ^~~~~~~
/usr/include/pb.h:478:5: note: in expansion of macro ‘SimpleMessage_FIELDLIST’
  478 |     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
      |     ^~~~~~~
simple.pb.c:9:1: error: ‘SimpleMessage’ {aka ‘struct _SimpleMessage’} has no member named ‘SimpleMessage’
    9 | PB_BIND(SimpleMessage, SimpleMessage, AUTO)
      | ^~~~~~~
/usr/include/pb.h:478:5: note: in expansion of macro ‘SimpleMessage_FIELDLIST’
  478 |     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
      |     ^~~~~~~
simple.pb.c:9:1: error: expression in static assertion is not constant
    9 | PB_BIND(SimpleMessage, SimpleMessage, AUTO)
      | ^
/usr/include/pb.h:478:5: note: in expansion of macro ‘SimpleMessage_FIELDLIST’
  478 |     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
      |     ^~~~~~~
```

I went with what other name for the first test field than the first letter of the alphabet: `required int32 a = 1;`

It turns out that this particular field name is not supported, apparently anything else is fine (at least I've tried all the other lowercase single letters). Just replace `lucky_number` with `a` in `examples/simple` and you'll get this error.

I'm using 0.4.1 as shipped by Ubuntu 20.04. Apologies for not testing the latest git, but at least I've taken a quick look at the changes and didn't spot anything that looked relevant.

Thanks!

PetteriAimonen, 2020-07-06:
I think this is fixed in 0.4.2, it was previously reported as issue #492.

egmontkob, 2020-07-06:
Oops, I missed this. Thanks a lot, and sorry for the noise!

closed 2020-07-06 by egmontkob