nanopb/nanopb #558: error in encoding proto2 message with option proto3:true and default != 0

mdionisio, 2020-07-17:
I have a message protobuf
`syntax = "proto2";

message pbA {
    optional float b = 2 [default=1.0];
    optional bool c = 3 [default=true];
}
`

and and options file like
`
pbA.b                   proto3:true
pbA.c                   proto3:true
`

the generator correctly does not generate has_b and has_c fields but the encoding does not work well because it does not honor the default field. 
 So if we encode a message with b=0.0 and c=false the b and c message is not encoded in the output stream because b=.0 and c=false is the correct default for protobuf3 but is if we decode that stream from another protobuf decoder that does not use the options files the result is not correct.

For me there is only 2 possible solutions:
1. the generator has to detect that proto3:true cannot be used with field that has a default value that does not match proto3 specification, and has to generate an error in compile time.
2. add a new HTYPE like PB_HTYPE_OPTIONAL_TRUE used by field with proto3:true and change encode_field(pb_ostream_t *stream, pb_field_iter_t *field) to always answer with true for this type (or use the real default to decide how answer)

PetteriAimonen, 2020-07-20:
Hmm, one option could be that encoder would check if the message description has default values, and in that case consider the field as always present. Because proto3 doesn't support default values, this would only happen with proto2 messages with `proto3` option defined manually. It can cause some fields to be encoded even when not needed (fields in message that has default values, but not on every field), but that is still better than current behavior.

mdionisio, 2020-07-20:
I agree with you. If you have a default value different from zero it is safer to encode the value in any case. But I don't know if you have duting the encoding phase the default value. For this I sugget to insert a new type PB_HTYPE_OPTIONAL_TRUE that is manage as PB_HTYPE_OPTIONAL during decoding and as REQUIRED during the encoding.

closed 2020-09-21 by PetteriAimonen