nanopb/nanopb #559: ImportError: attempted relative import with no known parent package

alexpacini, 2020-07-20:
In the compile phase, using CMake and make, I get:
```python
Traceback (most recent call last):
  File "[relative path hidden here]/build/nanopb/generator/nanopb_generator.py", line 42, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package
```

which is fixed by removing the `.` in nanopb_generator.py:42

```python
try:
    from proto import nanopb_pb2
    from proto._utils import invoke_protoc
```

I saw that it was recently converted to a module, so maybe having the `.` was intended.
However, this would mean that the call to the generator should be done from one directory above in the CMake.

PetteriAimonen, 2020-07-20:
Which version are you using?

alexpacini, 2020-07-20:
Just cloned from master.

I think that the call in the cmake should be using in some way `python -m` to run it as a module. Not sure how it will handle the path though.

PetteriAimonen, 2020-07-20:
Hmm, for some reason the error message does not appear to match the line 42 of newest master:
https://github.com/nanopb/nanopb/blob/master/generator/nanopb_generator.py

There is a fallback for ImportError that imports without `.`. It sounds like you might be running 0.4.0 generator for some reason, in which case the problem should be fixed in 0.4.1 and newer.

alexpacini, 2020-07-20:
Uhm, my submodule is at `e4f7b24`.
But then I am looking at the file that is placed my the build directory from the cmake script, which is basically using the FindNanopb in the extra folder (of that submodule).

alexpacini, 2020-07-20:
My fault. I did not notice that the repo I was working on had a second nanopb folder buried which used nanopb not as git submodule. I updated that folder to use submodules and all is working fine.
Apologies and thanks for your support.

closed 2020-07-20 by alexpacini