nanopb/nanopb #565: Warnings on "pb_decode.c" file, using the definition "PB_FIELD_32BIT" on file pb.h

andrealima, 2020-07-28:
For my project I need to use the definition "PB_FIELD_32BIT", as you can see below:
`/* Add support for tag numbers > 65536 and fields larger than 65536 bytes. */`
` #define PB_FIELD_32BIT 1`

But, the follows warnings are appointed on file "pb_decode.c" when I used the definition "PB_FIELD_32BIT":

**`Warning[Pa084]: pointless integer comparison, the result is always false <path>\pb_decode.c 1440`**
on function: `static bool checkreturn pb_dec_bytes(pb_istream_t *stream, const pb_field_t *field, void *dest)`
on line: 1440 `    if (size > PB_SIZE_MAX)
        PB_RETURN_ERROR(stream, "bytes overflow");`

**`Warning[Pa084]: pointless integer comparison, the result is always false <path>\pb_decode.c 1534`**
on function: `static bool checkreturn pb_dec_fixed_length_bytes(pb_istream_t *stream, const pb_field_t *field, void *dest)`
on line 1534:  `    if (size > PB_SIZE_MAX)
        PB_RETURN_ERROR(stream, "bytes overflow");`

I used some software that check this warnings, so I need to clean all warning of the project.
**Could I wait a fix about this warning?**

andrealima, 2020-07-28:
I have this warning on version packet "nanopb-0.3.9.6-windows-x86.zip"

PetteriAimonen, 2020-07-28:
Yeah, these are very difficult to avoid.

I don't think there will be a fix for this in the 0.3.x series, as it is in bugfix only maintenance now.
For 0.4.x maybe, but it's not a particularly high priority.

If the warning cannot be disabled in the checker, one workaround would be to change the `PB_SIZE_MAX` definition in pb.h to `(((pb_size_t)-1)-1)`, which would make the comparison no longer always false.

andrealima, 2020-07-28:
I'll share these information with my team.

We haven't updated yet because this note _"The x.x.9 releases are long-term-support"_.
With this note, we believe that the stable versions or the version that we can continue receiving support, should be version that follow this note.
If this conclusion is incorrect, please let me know!

Thank you for your quickly reply!

PetteriAimonen, 2020-07-28:
Yes, the long term support releases will receive bug fixes for the foreseeable future. But they won't receive new features, or fixes for low importance things like spurious compiler warnings if there is no actual bug.

andrealima, 2020-07-28:
Perfect, thanks for clarification.

closed 2020-07-28 by andrealima