nanopb/nanopb #566: Problems with generator on Ubuntu 20.04

pjorourke05, 2020-07-30:
I have done the following and have had no luck getting my proto files to generate on linux. Using the windows binaries I have no issues. dont know if I am missing a step but here is what I have done
1. Fresh VM of Ubuntu 20.04
2. Downloaded `nanopb-0.4.2-linux-x86.tar.gz`
3. installed `pip3`
4. ran `pip install protobuf grpcio-tools`
5. created a simple proto file that imported nanopb.proto
6. ran the generator using the binary `../nanopb-0.4.2-linux-x86/generator-bin/nanopb_generator -D $PWD/output -I $PWD/RADIANT -I ../nanopb-0.4.2-linux-x86/generator/proto command.proto errorCode.proto`

I get the following error

/home/radiant/Downloads/nanopb-0.4.2-linux-x86/generator-bin/proto: warning: directory does not exist.
/home/radiant/Downloads/nanopb-0.4.2-linux-x86/generator-bin/grpc_tools/_proto: warning: directory does not exist.
command.proto:9:1: warning: Import nanopb.proto is unused.
/home/radiant/Downloads/nanopb-0.4.2-linux-x86/generator-bin/proto: warning: directory does not exist.
/home/radiant/Downloads/nanopb-0.4.2-linux-x86/generator-bin/grpc_tools/_proto: warning: directory does not exist.
Writing to /home/radiant/Downloads/icp-scp/output/errorCode.pb.h and /home/radiant/Downloads/icp-scp/output/errorCode.pb.c
Writing to /home/radiant/Downloads/icp-scp/output/command.pb.h and /home/radiant/Downloads/icp-scp/output/command.pb.c
Traceback (most recent call last):
  File "nanopb_generator.py", line 2122, in <module>
  File "nanopb_generator.py", line 2021, in main_cli
  File "nanopb_generator.py", line 1956, in process_file
  File "nanopb_generator.py", line 1612, in generate_header
  File "nanopb_generator.py", line 1169, in fields_declaration
  File "nanopb_generator.py", line 1299, in default_value
  File "google/protobuf/descriptor.py", line 1077, in MakeDescriptor
TypeError: Couldn't build proto file into descriptor pool!
Invalid proto descriptor for file "d3674df1632142527303e84f693c037b.proto":
  NanoPBOptions.long_names: "NanoPBOptions.long_names" is already defined in file "nanopb.proto".
  NanoPBOptions.packed_struct: "NanoPBOptions.packed_struct" is already defined in file "nanopb.proto".
  NanoPBOptions.packed_enum: "NanoPBOptions.packed_enum" is already defined in file "nanopb.proto".
  NanoPBOptions.skip_message: "NanoPBOptions.skip_message" is already defined in file "nanopb.proto".
  NanoPBOptions.no_unions: "NanoPBOptions.no_unions" is already defined in file "nanopb.proto".
  NanoPBOptions.anonymous_oneof: "NanoPBOptions.anonymous_oneof" is already defined in file "nanopb.proto".
  NanoPBOptions.proto3: "NanoPBOptions.proto3" is already defined in file "nanopb.proto".
  NanoPBOptions.proto3_singular_msgs: "NanoPBOptions.proto3_singular_msgs" is already defined in file "nanopb.proto".
  NanoPBOptions.enum_to_string: "NanoPBOptions.enum_to_string" is already defined in file "nanopb.proto".
  NanoPBOptions.fixed_length: "NanoPBOptions.fixed_length" is already defined in file "nanopb.proto".
  NanoPBOptions.fixed_count: "NanoPBOptions.fixed_count" is already defined in file "nanopb.proto".
  NanoPBOptions.submsg_callback: "NanoPBOptions.submsg_callback" is already defined in file "nanopb.proto".
  NanoPBOptions.default_has: "NanoPBOptions.default_has" is already defined in file "nanopb.proto".
  NanoPBOptions.type_override: "NanoPBOptions.type_override" is already defined in file "nanopb.proto".
  NanoPBOptions.sort_by_tag: "NanoPBOptions.sort_by_tag" is already defined in file "nanopb.proto".
  NanoPBOptions: "NanoPBOptions" is already defined in file "nanopb.proto".

[3326] Failed to execute script nanopb_generator

Here is my top level proto file

```
syntax = "proto2";

import "errorCode.proto";
import "nanopb.proto";

message CommandMsg {
    required ErrorCode errCode = 1;
}
```
but have had no luck getting this to compile on linux.. I can compile this on windows just fine.. any help would be greatly appericated

PetteriAimonen, 2020-07-31:
Hmm, I think this may due to some unfortunate mistakes with the 0.4.2 Linux binary package.

I've already applied some fixes to the newest git master, can you try a package from here?
https://jpa.kapsi.fi/jenkins/job/nanopb/job/nanopb%20linux/

pjorourke05, 2020-07-31:
@PetteriAimonen  

Yes that package worked for me using the bin. I also tried to run it using the python script but had no luck this might be do my python setup dont know though.

`$python3 ./nanopb/generator/nanopb_generator.py -D $PWD/output -I $PWD/RADIANT -I $PWD/Injector -I $PWD/nanopb/generator/proto command.proto errorCode.proto radiantCommand.proto injCommand.proto`

I get the following error:
```
Traceback (most recent call last):
  File "./nanopb/generator/nanopb_generator.py", line 50, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./nanopb/generator/nanopb_generator.py", line 71, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/home/developer/Downloads/icp-scp/nanopb/generator/proto/nanopb_pb2.py", line 9, in <module>
    from google.protobuf import symbol_database as _symbol_database
ImportError: cannot import name 'symbol_database' from 'google.protobuf' (/home/developer/.local/lib/python3.8/site-packages/google/protobuf/__init__.py)
```
The reason I ask this is because I want use the cmake files to auto generate these when I build my application which it looks like I can do with extra/FindNanopb.cmake. My hunch though is this uses python and the python script not the generate binaries in generator-bin folder. Again any help would be much appreciated.

PetteriAimonen, 2020-07-31:
Yeah, I think that is what the CMake rules do. They've been written by various contributors and I'm not very familiar with them.

The error messages sound like some kind of version conflict between different protoc and/or python versions on your system. You could try removing `generator/proto/nanopb_pb2.py` and it should get regenerated automatically, possibly fixing the issue.

pjorourke05, 2020-07-31:
@PetteriAimonen 

Same result

Python Version: 3.8.2

Python packages list:
```
Package                Version      
---------------------- -------------
altgraph               0.17         
apturl                 0.5.2        
bcrypt                 3.1.7        
beautifulsoup4         4.9.1        
blinker                1.4          
bottle                 0.12.18      
Brlapi                 0.7.0        
certifi                2019.11.28   
chardet                3.0.4        
colorama               0.4.3        
command-not-found      0.3          
conan                  1.27.1       
conan-package-tools    0.34.0       
cryptography           2.8          
cupshelpers            1.0          
dbus-python            1.2.16       
defer                  1.0.6        
deprecation            2.0.7        
distro                 1.1.0        
distro-info            0.23ubuntu1  
duplicity              0.8.12.0     
entrypoints            0.3          
fasteners              0.14.1       
future                 0.18.2       
google                 3.0.0        
grpcio                 1.30.0       
grpcio-tools           1.30.0       
httplib2               0.14.0       
idna                   2.8          
Jinja2                 2.11.2       
keyring                18.0.1       
language-selector      0.1          
launchpadlib           1.10.13      
lazr.restfulclient     0.14.2       
lazr.uri               1.0.3        
lockfile               0.12.2       
louis                  3.12.0       
macaroonbakery         1.3.1        
Mako                   1.1.0        
MarkupSafe             1.1.0        
monotonic              1.5          
nanopb                 0.4.2        
netifaces              0.10.4       
node-semver            0.6.1        
oauthlib               3.1.0        
olefile                0.46         
packaging              20.4         
paramiko               2.6.0        
patch-ng               1.17.4       
pexpect                4.6.0        
Pillow                 7.0.0        
pip                    20.0.2       
pluginbase             0.7          
protobuf               3.12.4       
protobuf-setuptools    0.1.1        
pycairo                1.16.2       
pycups                 1.9.73       
Pygments               2.6.1        
PyGObject              3.36.0       
PyInstaller            3.6          
PyJWT                  1.7.1        
pymacaroons            0.13.0       
PyNaCl                 1.3.0        
pyparsing              2.4.7        
pyRFC3339              1.1          
python-apt             2.0.0        
python-dateutil        2.7.3        
python-debian          0.1.36ubuntu1
python3-protobuf       2.5.0        
pytz                   2019.3       
pyxdg                  0.26         
PyYAML                 5.3.1        
reportlab              3.5.34       
requests               2.22.0       
requests-unixsocket    0.2.0        
SecretStorage          2.3.1        
setuptools             45.2.0       
simplejson             3.16.0       
six                    1.14.0       
soupsieve              2.0.1        
systemd-python         234          
tabulate               0.8.7        
tqdm                   4.48.0       
ubuntu-advantage-tools 20.3         
ubuntu-drivers-common  0.0.0        
ufw                    0.36         
unattended-upgrades    0.1          
urllib3                1.25.8       
usb-creator            0.3.7        
wadllib                1.3.3        
wheel                  0.34.2       
xkit                   0.0.0 
```

```
python3 ./nanopb/generator/nanopb_generator.py -D $PWD/output -I $PWD/RADIANT -I $PWD/Injector command.proto errorCode.proto radiantCommand.proto injCommand.proto
Traceback (most recent call last):
  File "./nanopb/generator/nanopb_generator.py", line 50, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./nanopb/generator/nanopb_generator.py", line 71, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/home/radiant/Downloads/icp-scp-new/icp-scp/nanopb/generator/proto/nanopb_pb2.py", line 23, in <module>
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
```

pjorourke05, 2020-07-31:
I figured out my issue after running `pip3 install --upgrade protobuf` then the python scripts worked! I never will understand python haha!

PetteriAimonen, 2020-08-01:
You're not the only one - this has brought a lot of annoyance to a lot of people. It used to be so that getting matching `protoc` version and `python-protobuf` version was enough, but nowadays there is also the Python 2 vs. 3 confusion. And if that's not enough, some systems may have Python 3.6, 3.7 and 3.8 installed simultaneously :)

pjorourke05, 2020-08-01:
@PetteriAimonen 

One last question how do i build the src to generate the generator-bin folder. Is it just running build.py?

PetteriAimonen, 2020-08-01:
The binary package build scripts are in `tools` folder.

closed 2020-08-02 by pjorourke05

ionut-arm, 2021-02-01:
Hello,

I've just encountered this issue so leaving this comment for future reference in case anyone else bumps into it. I ran into the `from google.protobuf import symbol_database as _symbol_database` error and tried the `pip3 install --upgrade protobuf` mentioned above, but it didn't work. Turns out I also had the `python3-protobuf` pip package installed which was taking precedence over the `protobuf` package. Once I uninstalled it, everything was fine 💯