nanopb/nanopb #569: Message of Unknown Size defined in Separate File Broken When inside Oneof

Simon-Cochrane, 2020-08-07:
See for exact line of code/PR that broke that feature, it works on previous versions. https://github.com/nanopb/nanopb/commit/3ff6743173f12625b4c3d7fdaa4dd80a1448cad5#r41268754

PetteriAimonen, 2020-08-07:
Related: #415, #494

I guess the most reasonable approach would be to have `#ifdef __cplusplus`. Not sure if there is another way to go about it.

Of course, it would be nice to make messages always available at generation time, so if you have a specific scenario where you think the messages should be found but aren't, feel free to file that as a separate issue.

Simon-Cochrane, 2020-08-07:
I crafted a small example that reproduces the issue:

a.proto
```
syntax = "proto3";

package a;

message AUnknown {
  string data = 1;
}

message A {
  oneof data {
     AUnknown unknown = 1;
     int32 x = 2;
  }
}
```
b.proto
```
syntax = "proto3";

import "a.proto";

package b;

message B {
  oneof data {
     a.AUnknown unknown = 1;
     int32 x = 2;
  }
}
```

a.pb.h
```
/* Maximum encoded size of messages (where known) */
/* a_AUnknown_size depends on runtime parameters */
/* a_A_size depends on runtime parameters */
```

b.pb.h
```
/* Maximum encoded size of messages (where known) */
union b_B_data_size_union {char f1[(6 + a_AUnknown_size)]; char f0[11];};
#define b_B_size                                 (0 + sizeof(b_B_data_size_union))
```

This fails because AUnknown_size is not defined. Both files are available at generation time.

PetteriAimonen, 2020-09-10:
Should be fixed now.

I made the generator add `#if defined(AUnknown_size)` guard around the definition of `B_size`.

closed 2020-09-21 by PetteriAimonen