nanopb/nanopb #570: Calculations of maximum encoded message size are wrong for repeated elements

tjeske, 2020-08-07:
The size information in the generated nanopb header file seams to be not correct.

proto file:
```
message MyMessage {
    repeated uint32 a = 1;
}
```
options file:
```
MyMessage.a int_size:IS_8 max_count:XXX
```
I have tried several values for `max_count`. Field number + array length are 2 bytes. I would expect that each array element of type `uint8` needs max. 2 bytes but this seams to be not the case.
- `max_count:1`: size = 4 bytes  (expectation 4 bytes)
- `max_count:2`: size = 6 bytes  (expectation 6 bytes)
- `max_count:3`: size = 9 bytes  (expectation 8 bytes)
- `max_count:4`: size = 12 bytes (expectation 10 bytes)

PetteriAimonen, 2020-08-07:
This is because protobuf specification allows storing arrays as either packed or unpacked. The encoded size define is larger of the two.

https://github.com/nanopb/nanopb/blob/master/generator/nanopb_generator.py#L769

tjeske, 2020-08-09:
Thanks for the info (and for nanopb in general :)! What do you mean by packed/unpacked in the context of arrays? I have ran some tests with sample input and could not create protobuf messages which exceed my expected payload size using proto3. The generated struct type by nanopb is also not bigger than expected.

PetteriAimonen, 2020-08-09:
Protobuf binary format has [two ways to encode arrays](https://developers.google.com/protocol-buffers/docs/encoding#packed). 

Unpacked arrays typically take more space. Nanopb always encodes arrays as packed. But other encoders could, and the reasoning is that if you want to receive a message of type `MyMessage`, then `MyMessage_size` should be enough buffer space for it.

tjeske, 2020-08-10:
Ah, now I get it. Thx!

closed 2020-08-10 by tjeske