nanopb/nanopb #572: Allow separating a single field out of oneof, for better callback access

PetteriAimonen, 2020-08-12:
Related to #175 and others.

The `oneof` and `pb_callback_t` features have classically been largely incompatible.
Because the storage is shared, there is no place to store the callback pointers.

The `no_unions` option can be used to un-share the storage area of oneofs.
However, it has the downside that the struct size expands by the size of all the oneof fields.
It also generates separate `has_field` booleans instead of combined `which_oneof`.

A more seamless alternative could be to allow separating only the parts of the oneof that do contain callbacks, leaving the others overlapping. Something like:

<pre>
struct MyMessage {
   pb_size_t which_myoneof;
   struct {
       union {
            NoCallbackSubmsg1 submsg1;
            NoCallbackSubmsg2 submsg2;
       };
       CallbackSubmsg3 submsg3;
   } myoneof;
};
</pre>

Possibly the storage could also be shared if only one submessage contains callbacks.

Things to check:
- How decoder can figure out whether the storage overlaps.
- Which standards support anonymous unions, perhaps have to have option to give it a name.
- How default value initializers work with this.
- If a callback field is inside union, is it guaranteed to be initialized to zero when assigned `MyMessage_init_zero`?

kaiyuz, 2023-05-16:
@PetteriAimonen I made some changes to make the `oneof` and `pb_callback_t` compitable. I think it's convinient for developer to decode the related message. Here is the patch for `nanopb-0.4.6`. 
----------------------------------------------------------------------------------
From 0000000000000000000000000000000000000000 Mon Sep 17 00:00:00 2001
From: kaiyu zhang <kaiyuzhang@geotab.com>
Date: Tue, 16 May 2023 16:35:03 -0400
Subject: [PATCH] Add compatibility for 'oneof' and 'pb_callback_t'

---
 generator/nanopb_generator.py | 16 ++++++++++++++++
 pb.h                          | 12 ++++++++----
 pb_common.c                   | 29 +++++++++++++++++++++++------
 pb_common.h                   |  2 ++
 pb_decode.c                   |  1 +
 pb_decode.h                   |  1 +
 pb_encode.c                   |  3 ++-
 pb_encode.h                   |  1 +
 8 files changed, 54 insertions(+), 11 deletions(-)

diff --git a/generator/nanopb_generator.py b/generator/nanopb_generator.py
index 819b993..a82433f 100755
--- a/generator/nanopb_generator.py
+++ b/generator/nanopb_generator.py
@@ -997,6 +997,7 @@ class ExtensionField(Field):
         result += str(self.msg)
         result += self.msg.fields_declaration(dependencies)
         result += 'pb_byte_t %s_default[] = {0x00};\n' % self.msg.name
+        result += self.msg.fields_callback()
         result += self.msg.fields_definition(dependencies)
         result += 'const pb_extension_type_t %s = {\n' % self.fullname
         result += '    NULL,\n'
@@ -1312,8 +1313,12 @@ class Message(ProtoElement):
             if self.callback_function != 'pb_default_field_callback':
                 result += "extern bool %s(pb_istream_t *istream, pb_ostream_t *ostream, const pb_field_t *field);\n" % self.callback_function
             result += "#define %s_CALLBACK %s\n" % (self.name, self.callback_function)
+            result += "#define %s_ENCODE_CALLBACK_FOR_ONEOF pb_%s_encode_callback_for_oneof\n" % (self.name, str(self.name).lower())
+            result += "#define %s_DECODE_CALLBACK_FOR_ONEOF pb_%s_decode_callback_for_oneof\n" % (self.name, str(self.name).lower())
         else:
             result += "#define %s_CALLBACK NULL\n" % self.name
+            result += "#define %s_ENCODE_CALLBACK_FOR_ONEOF NULL\n" % self.name
+            result += "#define %s_DECODE_CALLBACK_FOR_ONEOF NULL\n" % self.name
 
         defval = self.default_value(dependencies)
         if defval:
@@ -1341,6 +1346,16 @@ class Message(ProtoElement):
         result += '};'
         return result
 
+    def fields_callback(self):
+        '''Return callback functions in this message.'''
+        result = ''
+        has_callbacks = bool([f for f in self.fields if f.has_callbacks()])
+        if has_callbacks:
+            result = "__attribute__((weak)) bool pb_%s_encode_callback_for_oneof(pb_ostream_t * stream, const pb_field_t * field, void * const *arg){return true;}\n" % str(self.name).lower()
+            result += "__attribute__((weak)) bool pb_%s_decode_callback_for_oneof(pb_istream_t * stream, const pb_field_t * field, void ** arg){return true;}\n" % str(self.name).lower()
+            result += "\n\n"
+        return result
+ 
     def fields_definition(self, dependencies):
         '''Return the field descriptor definition that goes in .pb.c file.'''
         width = self.required_descriptor_width(dependencies)
@@ -1929,6 +1944,7 @@ class ProtoFile:
         yield '\n'
 
         for msg in self.messages:
+            yield msg.fields_callback()
             yield msg.fields_definition(self.dependencies) + '\n\n'
 
         for ext in self.extensions:
diff --git a/pb.h b/pb.h
index f5c9c6f..94baa8e 100644
[... 142 lines cut ...]
+    stream.callback_arg_for_oneof = NULL;
     return stream;
 }
 
@@ -345,7 +346,7 @@ static bool checkreturn pb_check_proto3_default_value(const pb_field_iter_t *fie
         else if (field->descriptor->field_callback == pb_default_field_callback)
         {
             pb_callback_t *pCallback = (pb_callback_t*)field->pData;
-            return pCallback->funcs.encode == NULL;
+            return (pCallback->funcs.encode == NULL) && (field->descriptor->field_encode_callback_for_oneof == NULL);
         }
         else
         {
diff --git a/pb_encode.h b/pb_encode.h
index 8913683..0b5724b 100644
--- a/pb_encode.h
+++ b/pb_encode.h
@@ -44,6 +44,7 @@ struct pb_ostream_s
 #ifndef PB_NO_ERRMSG
     const char *errmsg;
 #endif
+    void *callback_arg_for_oneof;
 };
 
 /***************************

PetteriAimonen, 2023-05-17:
@kaiyuz That is an interesting approach. I'm not sure if it is the best approach in general, I think it will still be difficult when there are multiple nested messages with oneofs.
Having the field callback in the stream (where you have callback argument) is an interesting option also.