nanopb/nanopb #576: #define PB_NO_ERRMSG causes buffer overflow when calling pb_encode.

tombull, 2020-08-30:
I was trying to save program space in my microcontroller firmware using nanopb. I decided to try the `#define PB_NO_ERRMSG` option. It seemed to work - except it only saved 24 bytes of program space. Then I started to get parts of my memory overwritten after calling `pb_encode`. There seems to be a buffer overflow in `pb_encode` if the `#define PB_NO_ERRMSG` option is set. 

I've not explored the code to find out where or why the buffer overflow is caused - but all my usage of `pb_decode` was fine - it did not cause any problems. Only when I called `pb_encode` was there a problem.

PetteriAimonen, 2020-08-30:
That sounds like `pb_encode.c` might have been compiled with different options than your own code that calls `pb_ostream_from_buffer()`.

Make sure you've recompiled all files, and that the `#define PB_NO_ERRMSG` option is defined so that it is visible in all compilation units, e.g. either in `pb.h` or on compiler command line.

tombull, 2020-08-30:
I was using a custom stream which writes directly to a websocket... but it also happens when using `pb_ostream_from_buffer`.

Defining `-DPB_NO_ERRMSG` on the compiler command line fixes the problem.

closed 2020-09-21 by PetteriAimonen