nanopb/nanopb #578: The mangle_names option doesn't work if project structure conforms to Google's Protobuf Style Guide

EvanShenkman-Sonos, 2020-09-01:
_nanopb version=4.2.0_
related to: #552 

I'm having trouble getting the `mangle_names` option to work in my project which follows Google's [style suggestion regarding packages](https://developers.google.com/protocol-buffers/docs/style#packages). It seems like **none** of the `TypenameMangling` options have any effect on the generated code given my project structure.

Any help would be greatly appreciated. I've broken down my project structure and contents below.

---

<details>
<summary>Expand to see project contents...</summary>

#### Project contents...

```bash
➜ tree .
.
└── foo
    └── bar
        └── v1
            ├── animal.options
            └── animal.proto

3 directories, 2 files
```

</details>

<details>
<summary>Expand to view the proto file...</summary>

#### Proto contents

```proto
➜ cat foo/bar/v1/animal.proto 
syntax = "proto3";

package foo.bar.v1;

message Animal {
    Species type = 1;
    string name = 2;
}

enum Species {
    SPECIES_UNSPECIFIED = 0;
    SPECIES_CAT = 1;
    SPECIES_DOG = 2;
};
```

</details>

<details>
<summary>Expand to view the options file...</summary>

#### Options contents

```bash
➜ cat foo/bar/v1/animal.options
foo.*                           mangle_names:M_STRIP_PACKAGE
foo.bar.v1.Animal.name          max_size:40
foo.bar.v1.Species              enum_to_string:true
foo.bar.v1.Species              long_names:false
```

</details>

<details>
<summary>Expand to see how I'm invoking protoc...</summary>

#### Invoking `protoc`

```bash
➜ mkdir output
➜ protoc \
    --plugin=protoc-gen-nanopb=/nanopb/generator-bin/protoc-gen-nanopb \
    --nanopb_out="-v -f foo/bar/v1/animal.options:output" \
    $(find . -name '*.proto') 
Google Python protobuf library imported from /nanopb/generator-bin/google/protobuf/__init__.pyc, version 3.12.2
Reading options from ./foo/bar/v1/animal.options
Options for foo/bar/v1/animal.proto: proto3: true

Options for foo.bar.v1.Species: long_names: false
proto3: true
enum_to_string: true
mangle_names: M_STRIP_PACKAGE

Options for foo.bar.v1.Animal: proto3: true
mangle_names: M_STRIP_PACKAGE

Options for foo.bar.v1.Animal.type: proto3: true
mangle_names: M_STRIP_PACKAGE

Options for foo.bar.v1.Animal.name: max_size: 40
proto3: true
mangle_names: M_STRIP_PACKAGE

Reading options from ./foo/bar/v1/animal.options
Options for foo/bar/v1/animal.proto: proto3: true

Options for foo.bar.v1.Species: long_names: false
proto3: true
enum_to_string: true
mangle_names: M_STRIP_PACKAGE

Options for foo.bar.v1.Animal: proto3: true
mangle_names: M_STRIP_PACKAGE

Options for foo.bar.v1.Animal.type: proto3: true
mangle_names: M_STRIP_PACKAGE

Options for foo.bar.v1.Animal.name: max_size: 40
proto3: true
mangle_names: M_STRIP_PACKAGE
```
</details>

<details>
<summary>Expand to see the resulting header file...</summary>

```c
➜ cat output/foo/bar/v1/animal.pb.h
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.2 */

#ifndef PB_FOO_BAR_V1_FOO_BAR_V1_ANIMAL_PB_H_INCLUDED
#define PB_FOO_BAR_V1_FOO_BAR_V1_ANIMAL_PB_H_INCLUDED
#include <pb.h>

#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif

#ifdef __cplusplus
extern "C" {
#endif

/* Enum definitions */
typedef enum _foo_bar_v1_Species {
    foo_bar_v1_SPECIES_UNSPECIFIED = 0,
    foo_bar_v1_SPECIES_CAT = 1,
    foo_bar_v1_SPECIES_DOG = 2
} foo_bar_v1_Species;

/* Struct definitions */
typedef struct _foo_bar_v1_Animal {
    foo_bar_v1_Species type;
    char name[40];
} foo_bar_v1_Animal;

/* Helper constants for enums */
#define _foo_bar_v1_Species_MIN foo_bar_v1_SPECIES_UNSPECIFIED
#define _foo_bar_v1_Species_MAX foo_bar_v1_SPECIES_DOG
#define _foo_bar_v1_Species_ARRAYSIZE ((foo_bar_v1_Species)(foo_bar_v1_SPECIES_DOG+1))
#define foo_bar_v1_Species_SPECIES_UNSPECIFIED foo_bar_v1_SPECIES_UNSPECIFIED
#define foo_bar_v1_Species_SPECIES_CAT foo_bar_v1_SPECIES_CAT
#define foo_bar_v1_Species_SPECIES_DOG foo_bar_v1_SPECIES_DOG
const char *foo_bar_v1_Species_name(foo_bar_v1_Species v);

/* Initializer values for message structs */
#define foo_bar_v1_Animal_init_default           {_foo_bar_v1_Species_MIN, ""}
#define foo_bar_v1_Animal_init_zero              {_foo_bar_v1_Species_MIN, ""}

/* Field tags (for use in manual encoding/decoding) */
#define foo_bar_v1_Animal_type_tag               1
#define foo_bar_v1_Animal_name_tag               2

/* Struct field encoding specification for nanopb */
#define foo_bar_v1_Animal_FIELDLIST(X, a) \
X(a, STATIC,   SINGULAR, UENUM,    type,              1) \
X(a, STATIC,   SINGULAR, STRING,   name,              2)
#define foo_bar_v1_Animal_CALLBACK NULL
#define foo_bar_v1_Animal_DEFAULT NULL

extern const pb_msgdesc_t foo_bar_v1_Animal_msg;

/* Defines for backwards compatibility with code written before nanopb-0.4.0 */
#define foo_bar_v1_Animal_fields &foo_bar_v1_Animal_msg

/* Maximum encoded size of messages (where known) */
#define foo_bar_v1_Animal_size                   43

#ifdef __cplusplus
} /* extern "C" */
#endif

#endif
```

</details>

---

I would expect that the package prefix, `foo_bar_v1_`, would be removed *if* the `M_STRIP_PACKAGE` option is given, however that is not the case in this scenario. The `protoc-gen-nanopb` plugin is clearly picking up that the option is specified, but the resulting source code does not reflect that.

PetteriAimonen, 2020-09-02:
The issue seems to be that the `mangle_names` option applies on the file level.

From the log:

     Options for foo/bar/v1/animal.proto: proto3: true

For me, it seems to work if I change the first line of `animal.options` to:

     *                           mangle_names:M_STRIP_PACKAGE

This is probably at least a documentation issue, though I'm not sure where it would be best to document it. The documentation regarding many of the options is quite lacking, mangle_names is not even mentioned at all.

EvanShenkman-Sonos, 2020-09-02:
@PetteriAimonen, that did the trick. Just to make sure I understand what you mean by...

> _... the mangle_names option applies on the file level._

The nanopb generator only applies the mangle_names option ***if*** it is specified as a top-level glob?

PetteriAimonen, 2020-09-02:
On code side it works so that when the generator starts processing a file, it checks for mangle_names option, but then does not check it again for each message.

Currently options for files are matched by their name, so `foo/bar/v1/animal.proto` or `foo/*` would work equally well as patterns.

I guess it could be reasonable to match by package name also, but I'm not sure if there are use cases that wouldn't work equally well by matching by file name.

EvanShenkman-Sonos, 2020-09-02:
Ah, I understand. I think there's just some messaging/communication issues in the documentation and in-code comments. There's no mention of the ability to specify options via file name in a `.options` file. In fact, the [documentation](https://jpa.kapsi.fi/nanopb/docs/reference.html#defining-the-options-in-a-options-file) implies that specifying file names is not valid.

> * All other lines should start with a field name pattern, followed by one or more options. For example: "**MyMessage.myfield max_size:5 max_count:10**".

I'm happy to spend some time over the next week or so to do some minor documentation cleanup if you'd like and thanks again for the help!

PetteriAimonen, 2020-09-02:
I made some very small fixes now, but if you find anything further that could help, please send a pull request :)

In addition to having a lack of time to dedicate to documenting nanopb, it is really difficult to put oneself into the position of a person not already knowing the details.

closed 2020-09-21 by PetteriAimonen