nanopb/nanopb #579: having trouble to pack repeated data

Donghua-Gu-CardinalPeak, 2020-09-03:
I am having trouble to pack repeated data, it looks like the pb_encode simply ignores the array. I have this problem with 0.3.9 and I just tried it on 0.4.2 and I have problem with both version. Can someone take a look? 
Here is modified test code based on the windows version:

**simple.proto:**
`syntax = "proto3";
option java_generate_equals_and_hash = true;
option optimize_for = CODE_SIZE;

message OtaLog_PB
{
	uint32 version_num                              = 1;
	uint32 num_of_ota_events                        = 2;
	repeated uint32 ota_log_history                 = 3; 
}
`

**simple.options:**
`
OtaLog_PB.ota_log_history  max_count:80`

**simple.c:**

`#include <stdio.h>
#include "pb_encode.h"
#include "pb_decode.h"
#include "simple.pb.h"

int main()
{
    /* This is the buffer where we will store our message. */
    uint8_t buffer1[512];
    size_t message_length1;
    bool status1;
    
    /* Encode our message */
    {
        /* Allocate space on the stack to store the message data.
         *
         * Nanopb generates simple struct definitions for all the messages.
         * - check out the contents of simple.pb.h!
         * It is a good idea to always initialize your structures
         * so that you do not have garbage data from RAM in there.
         */
        OtaLog_PB message = OtaLog_PB_init_zero;

        /* Create a stream that will write to our buffer. */
        pb_ostream_t stream1 = pb_ostream_from_buffer(buffer1, sizeof(buffer1));

        /* Fill in the lucky number */
        message.version_num = 1;
        message.num_of_ota_events = 21;
    
        for (int i = 0; i < 80; i++)
        {
            message.ota_log_history[i] = i * 101;
        }
        
        /* Now we are ready to encode the message! */
        status1 = pb_encode(&stream1, OtaLog_PB_fields, &message);
        message_length1 = stream1.bytes_written;
        printf("message length is %lu\n\r", message_length1);
        /* Then just check for any errors.. */
        if (!status1)
        {
            printf("Encoding failed: %s\n\r", PB_GET_ERROR(&stream1));
            return 1;
        }
        
        printf("protobuf raw:\n\r");
        int j = 0;
        for (int i = 0 ; i < message_length1; i++)
        {
            printf(" %02X ", buffer1[i]);
            j++;
            if (j == 8)
            {
                printf("\n\r");
                j = 0;
            }
        }
        printf("\n\r");
    }
    
    /* Now we could transmit the message over network, store it in a file or
     * wrap it to a pigeon's leg.
     */
    printf("----------------------------\n\r");
    /* But because we are lazy, we will just decode it immediately. */
    {
        /* Allocate space for the decoded message. */
        OtaLog_PB message = OtaLog_PB_init_zero;

        /* Create a stream that reads from the buffer. */
        pb_istream_t stream1 = pb_istream_from_buffer(buffer1, message_length1);

        /* Now we are ready to decode the message. */
        status1 = pb_decode(&stream1, OtaLog_PB_fields, &message);
        printf("after reading stream, length is %lu\n\r",message_length1);

        /* Check for errors... */
        if (!status1)
        {
            printf("Decoding failed: %s\n\r", PB_GET_ERROR(&stream1));
            return 1;
        }
        
        /* Print the data contained in the message. */
        printf("version is %d!\n\r", message.version_num);
        printf("ota event numb is %d!\n\r", message.num_of_ota_events);
        int i = 0;
        while(message.ota_log_history[i] != 0)
        {
            printf( "%2x", message.ota_log_history[i++]);
        }
        printf("----------------------------\n\r");
    }
    
    return 0;
}
`

Donghua-Gu-CardinalPeak, 2020-09-03:
And I am confused about field.pSize, is this linked to number of the elements in the array? Why is it a boolean value? In the encode_array function, because field->pSize is 0, the function simply return true without doing anything.

PetteriAimonen, 2020-09-04:
Take a look inside the `.pb.h` file - there should be a `pb_size_t ota_log_history_count;` field there.
Set that to the number of entries in your array.

`pSize` is a pointer to the size field, which is either `bool has_field;` for proto2-syntax optional fields, or `pb_size_t field_count;` for repeated fields.

Donghua-Gu-CardinalPeak, 2020-09-06:
got it working. thank you

closed 2020-09-06 by Donghua-Gu-CardinalPeak