nanopb/nanopb #583: issue when migrating from 0.3.9 to 0.4.2

Donghua-Gu-CardinalPeak, 2020-09-08:
Our existing code base uses version 0.3.9, and I am trying to upgrade it to 0.4.2. However, in the generated pb.h file, under maximum encoded size of message, there is something new:
`union ControlEnvelope_payload_size_union {char f9[(8 + sizeof(Response_payload_size_union))]; char f10[(12 + Dialog_size)]; char f11[(26 + SpeechSettings_size + SpeechInitiator_size + Dialog_size)]; char f12[(14 + Dialog_size)]; char f13[(12 + Dialog_size)]; char f25[(13 + DeviceConfiguration_size)]; char f101[(13 + State_size)]; char f102[(13 + State_size)]; char f0[38];};`

The compiler reports Error 'Response_payload_size_union' undefined. 
0.3.9 doesn't have anything under maximum encoded size, so do I need to manually create this union somewhere else or the nanopb supposes to auto-generate this?
Thanks,

PetteriAimonen, 2020-09-08:
Sounds like issue #569.

You can try passing all `.proto` files to the generator at the same time, to try to make sure it finds them and doesn't have to resort to the compile time calculation of message sizes that is causing the problem.

Donghua-Gu-CardinalPeak, 2020-09-08:
this particular proto file import 11 other proto files, so is it just the sequence of the files generation is out of order? so ControlEnvelope_payload_size_union is passed to generator before Response_payload is passed to generator? 
Do I need to combine all the proto into one file or re-arrange my build script to make sure response.proto is built first before controlEnvelope.proto?

PetteriAimonen, 2020-09-09:
You could try something like  `protoc --nanopb_out=. response.proto controlEnvelope.proto` with all the proto files listed. The order does not matter, but having them in the same command (instead of 11 separate commands) ensures that the generator finds them.

Donghua-Gu-CardinalPeak, 2020-09-10:
Thank you, it works.

closed 2020-09-10 by Donghua-Gu-CardinalPeak