nanopb/nanopb #588: [Question] Are complex encoding of nested repeated submessages supported?

dnuch, 2020-09-22:
For example:

```
syntax = "proto3";

message line {
  string value = 2;
}

message info {
  repeated line lines = 3;
}

message log {
  repeated info infos = 2;
}
```

I can encode message info and line just fine but log has been problematic. My apologies if this has been answered elsewhere but I can't find an example this complex.

Thanks.

PetteriAimonen, 2020-09-22:
There are many ways to do this: static allocation, callbacks and malloc(). To figure out what is best for your case, I'll need a bit more information:

- Do you know the maximum number of characters per line, lines per info or infos per log in advance?
- What will you do with the lines once they are decoded?

PetteriAimonen, 2020-09-22:
Oh, or are you just encoding? But in that case the same questions apply: do you know the maximum sizes, and where do the lines come from?

dnuch, 2020-09-24:
I ended up getting it to work! Sorry for the noise

closed 2020-09-24 by dnuch