nanopb/nanopb #589: Problem with IAR Compiler; filed struct not generated and 64 Bit Txpes possibly not supported

krisnov1, 2020-09-22:
Hi folks,
I have a problem.
 I started generation C code without a problem.
The first compilation problem occured here: pb.h

**If you get errors here, it probably means that your stdint.h is not corect for your platform.
#ifndef PB_WITHOUT_64BIT
PB_STATIC_ASSERT(sizeof(int64_t) == 2 * sizeof(int32_t), INT64_T_WRONG_SIZE)
PB_STAT**
```
I fixed that by using the switch **PB_WITHOUT_64BIT**. Hope that is OK

The second compilation problem occured here: simple.pb.c 

Error[Pe077]: this declaration has no storage class or type specifier C:\Users\OXF_KNO\work\blueprint-package-integration-test\tpt_test\Nucleo_Node\Nucleo_Cfg\simple.pb.c 9 
"PB_BIND(SimpleMessage, SimpleMessage, AUTO)"

I figured out the structure was not generated. Don’t know why
I was abel to “fix” that problem by using the switch **#define PB_NO_STATIC_ASSERT ////TODO KN?**
 

  buffer = (8, 1, 16, 2, 24, 3, 145, 1, 24, 180, 164, 224, 179, 5, 0, 0, 0, 0, 0,
  message = (
    lucky_number1 = 0x1,
    lucky_number2 = 0x2,
    lucky_number3 = 0x3)

 
The data gets written and retrieved as expected. BUT the pb_decode returns an error status.

What am I doing wrong here..
Regards

Krisitjan

PetteriAimonen, 2020-09-23:
It does sound a lot like the compiler or it's preprocessor is not standard-compliant enough to handle the macro tricks that nanopb uses to generate the descriptor for the structures.

Are you able to upload the preprocessed output of the .pb.h file and corresponding .pb.h?

krisnov1, 2020-09-23:
[List.zip](https://github.com/nanopb/nanopb/files/5266482/List.zip)
The seems to be no pb.h (preprocessed output ). But I found structures in simple.pb.i. I send all the preprocessed output to make sure you are not missing something .

PetteriAimonen, 2020-09-24:
Hmm, it sounds like the main problem could be that `PB_STATIC_ASSERT` macro is not working correctly on IAR for some reason. I don't have that compiler myself, so it is pretty difficult to troubleshoot.

It looks like the compiler should support 64-bit types just fine, it is just that the STATIC_ASSERT is failing for some reason.

Based on the files you've sent, the compiler is currently set to C89 standard mode. Could you try setting it to C11 standard or newer, because that would enable the standard `_Static_assert()` mechanism that might work better.

--

Other than that it being nicer to get notified of problems at compile time, it should work with `PB_NO_STATIC_ASSERT`.
You could check the stream error message when `pb_decode()` returns an error.

krisnov1, 2020-10-02:
[image]
To my surprise the compiler showed C11 already. But I changed some other setttings. Than it was possible to compile the whole code without disabling any options. For me it seems to be OK now. Also the decode function rerurns with no errror code now.
Thanks for the support!

closed 2020-10-02 by PetteriAimonen

arzippi, 2020-11-27:
Hi @krisnov1,

I ran into the exact same problem with the IAR compiler.
Can you describe which compiler settings you had to change? My compiler is already set to C11.

For now I use **PB_WITHOUT_64BIT** as a workaround...

krisnov1, 2020-11-30:
Hi Arzippi I used exactly the settings as published in the screenshot above. an this has solved my problems.

arzippi, 2020-11-30:
Thanks for your reply @krisnov1!

My compiler language settings look exactly the same. The only difference is that I use extension-based, because I have some C++ code.
[image]
I'm using the IAR Eclipse plugin, but the compiler configuration should be identical.

I already tried looking into the other compiler settings, but there was nothing that seemed suspicious... Maybe there is something incompatibel between this version of the IAR compiler and the STATIC_ASSERT of the newer nanopb versions...

krisnov1, 2020-12-02:
I am not evry experienced with the IAR. It is possible that different compiler versions behave different with the same settings..

ghseb, 2021-06-15:
Observed the same issue. To me it seems that IAR compiler has a `static_assert` keyword but no `_Static_assert` definition. See `static_assert` at `IAR C language extensions` in https://wwwfiles.iar.com/v850/webic/ew/doc/EWV850_CompilerReference.pdf

PetteriAimonen, 2021-06-15:
@ghseb Would you want to make a pull request to `pb.h` to make it use `static_assert` for IAR? I can't test such change myself.