nanopb/nanopb #592: Error buil with external enum

ersivv, 2020-09-27:
I connect an enumeration from an external file
```proto
// mode.proto
syntax = "proto2";
package pb;

enum ModeType {
  MANUAL = 0;
  AUTO = 1;
}
```

```proto
// test.proto
syntax = "proto2";
import "mode.proto";
package pb;

message Test {
  required ModeType mode = 1;
}
```

getting error while building:

```python
python .\nanopb\generator\nanopb_generator.py --options-path=protos --verbose .\protos\test.proto
Google Python protobuf library imported from c:\Python\lib\site-packages\google\protobuf\__init__.py, version 3.13.0
Options file not found: test.options
Options file not found: test.options
Options for test.proto:
Options for pb.Test:
Options for pb.Test.mode:
Traceback (most recent call last):
  File ".\nanopb\generator\nanopb_generator.py", line 2153, in <module>
    main_cli()
  File ".\nanopb\generator\nanopb_generator.py", line 2049, in main_cli
    results = process_file(fdesc.name, fdesc, options, other_files)
  File ".\nanopb\generator\nanopb_generator.py", line 1984, in process_file
    headerdata = ''.join(f.generate_header(includes, headerbasename, options))
  File ".\nanopb\generator\nanopb_generator.py", line 1631, in generate_header
    yield msg.fields_declaration(self.dependencies) + '\n'
  File ".\nanopb\generator\nanopb_generator.py", line 1186, in fields_declaration
    defval = self.default_value(dependencies)
  File ".\nanopb\generator\nanopb_generator.py", line 1294, in default_value
    enumtype = dependencies[str(enumname)]
KeyError: 'pb_ModeType'
```

PetteriAimonen, 2020-09-27:
Hmm, yeah, that should work even without the enum being available.

You could pass all files to `nanopb_generator.py` at once, then it should find the dependency:

    python .\nanopb\generator\nanopb_generator.py --options-path=protos --verbose .\protos\test.proto .\protos\mode.proto

ersivv, 2020-09-27:
Yes, that is how the build was successful. Thanks.

PetteriAimonen, 2020-09-27:
I'll keep this open to add some better fallback in case the file is actually not available.

ersivv, 2020-09-27:
The problem is different, the file is seen through import and without specifying an external one.
For example, if you do this:
```proto
// mode.proto
syntax = "proto2";
package pb;

enum ModeType {
  MANUAL = 0;
  AUTO = 1;
}
```

```proto
// test.proto
syntax = "proto2";
import "mode.proto";
package pb;

enum ModeType {
  MANUAL = 0;
  AUTO = 1;
}

message Test {
  required ModeType mode = 1;
}
```

I get an error already defined:

```cmd
python .\nanopb\generator\nanopb_generator.py --options-path=protos --verbose .\protos\test.proto
Google Python protobuf library imported from c:\Python\lib\site-packages\google\protobuf\__init__.py, version 3.13.0
test.proto:7:3: "pb.MANUAL" is already defined in file "mode.proto".
test.proto:7:3: Note that enum values use C++ scoping rules, meaning that enum values are siblings of their type, not children of it.  Therefore, "MANUAL" must be unique within "pb", not just within "ModeType".
test.proto:8:3: "pb.AUTO" is already defined in file "mode.proto".
test.proto:8:3: Note that enum values use C++ scoping rules, meaning that enum values are siblings of their type, not children of it.  Therefore, "AUTO" must be unique within "pb", not just within "ModeType".
test.proto:6:6: "pb.ModeType" is already defined in file "mode.proto".
```

 That is, the import works. Doesn't work if you import only the enumeration.

PetteriAimonen, 2020-11-25:
Looks like it is not possible to correctly generate the default values when the dependency file is not found.
I ended up just adding a better error message in this case.

closed 2020-11-25 by PetteriAimonen