nanopb/nanopb #594: RHEL support

lecndav, 2020-09-28:
Hi,

the git clone command on my RHEL (v7.7) machine results in 

```
fatal: unable to access 'https://android.googlesource.com/platform/external/nanopb-c/': Invalid file descriptor
```

Is RHEL not supported or is there a way to resolve this issue?

PetteriAimonen, 2020-09-28:
Sounds like there is some kind of problem with your git client and/or the android.googlesource.com repository. [This issue report](https://github.com/ansible/awx/issues/5052) suggests that removing the trailing slash from the URL may help.

lecndav, 2020-09-28:
I'm able to clone other repositories successfully. Removing the trailing slash didn't help either. I just updated my git client (now v1.8.3.1) but the problem still persists.

PetteriAimonen, 2020-09-28:
Isn't git 1.8.3 from 2013 or something? But anyway, not much I can do about it as I do not control the android.googlesource.com repository. This does not seem to be a problem with nanopb, just a problem with either your git client or the server.

lecndav, 2020-09-28:
Alright, I try to manually update git to 2.x because it is not provided in our repository.

Thanks for you support anyway.

closed 2020-09-28 by lecndav