nanopb/nanopb #597: Make count parameter of stream (en-/de-)coding callbacks in/out

shiirish, 2020-10-01:
**What should the feature do?**
Allow stream encoding callbacks to return how many bytes were actually written to the stream so that they may differ from those requested. This would allow users of nanopb to encode for a given protocol directly to a buffer.

**In what situation would the feature be useful?**
For protocols such as SLIP encoding, a character written to a SLIP frame may need to be preceded with an escape character. Meaning one byte may actually encode as two bytes to the buffer. If the *count* parameter were an in/out pointer to type *size_t*, the stream encoding callback could return how many bytes were actually written to the stream. Subsequent calls to pb_write would have a correct *bytes_written* reported, and *stream full* errors would be reported correctly.

E.g., the change to pb_write may be as simple as the following in *pb_encode.h*:
```C
bool (*callback)(pb_ostream_t *stream, const pb_byte_t *buf, size_t *count);
```

And invoking the callback would change to:
```C
bool checkreturn pb_write(pb_ostream_t *stream, const pb_byte_t *buf, size_t count)
{
    if (count > 0 && stream->callback != NULL)
    {
        if (stream->bytes_written + count < stream->bytes_written ||
            stream->bytes_written + count > stream->max_size)
        {
            PB_RETURN_ERROR(stream, "stream full");
        }

#ifdef PB_BUFFER_ONLY
        if (!buf_write(stream, buf, count))
            PB_RETURN_ERROR(stream, "io error");
#else        
        if (!stream->callback(stream, buf, &count))
            PB_RETURN_ERROR(stream, "io error");
#endif
    }
    
    stream->bytes_written += count;
    return true;
}
```

PetteriAimonen, 2020-10-01:
Hmm, I think it would be a bit more complex because `bytes_written` is used in `pb_encode_submessage()` to verify that the submessage size stays the same after it has been calculated. And on decoding side in `pb_open_string_substream()`.

I've been thinking about some kind of refactoring for the stream interface, but haven't figured out a good design so far. It would be great to be able to encode at least short lengths to a memory buffer instead of calling the callback for each byte.

shiirish, 2020-10-01:
Thanks for replying, @PetteriAimonen. I appreciate your time.

I think I understand the issue regarding *pb_encode_submessage()*. The assumed encoding length for a sub-message is determined before the real encoding callback is invoked. Then after actually encoding the encoded length is checked against the assumed length. Is that correct?

Also, I'm not sure I understand your meaning in your second paragraph; I don't think you're referencing the call to the stream's callback function in *pb_write()*, because that is not a byte-by-byte call to that callback. Could you elaborate?

Going forward with this project today, I think I will have to roughly triple my buffer space to solve this problem. Messages will be encoded for protobuf to one buffer, then that buffer will be SLIP encoded to another buffer before being transmitted via the UART using DMA.

Previously, I would encode the protobuf data then SLIP encode while writing each byte to the serial port. Timing constraints now dictate that CPU cycles not be spent waiting for the UART to ship each byte. If I were able to encode for SLIP directly in the callback, I could avoid this issue.

PetteriAimonen, 2020-10-02:
As for doing this today without changes to nanopb, you could have a separate position variable for yourself, independent of the `bytes_written`.

This test case may be helpful as a template, though it is for input and not output stream:
https://github.com/nanopb/nanopb/blob/master/tests/fuzztest/flakystream.h
https://github.com/nanopb/nanopb/blob/master/tests/fuzztest/flakystream.c

I'm thinking something like this:

<pre>
typedef struct {
    pb_ostream_t stream;
    const uint8_t *buffer;
    size_t position;
    size_t buflen;
} slipstream_t;

bool slipstream_callback(pb_ostream_t *stream, const pb_byte_t *buf, size_t count)
{
    slipstream_t *state = stream->state;
    /* Encode to state->buffer[state->position++] and check that state->buflen is not exceeded. */
}

void slipstream_init(slipstream_t *state, uint8_t *buffer, size_t buflen)
{
   memset(stream, 0, sizeof(*stream));
   state->stream.callback = slipstream_callback;
   state->stream.max_size = SIZE_MAX;
   state->stream.state = state;
   state->buffer = buffer;
   state->position = 0;
   state->buflen = buflen;
}

void main()
{
    slipstream_t slipstate;
    slipstream_init(&slipstate, uart_buffer, sizeof(uart_buffer));
    pb_encode(&slipstate->stream, MyMsg_fields, &msg);
}
</pre>

That way nanopb doesn't see or care about the size of the underlying buffer.

shiirish, 2020-10-02:
That's a great solution, thanks for that! And thanks for putting the effort into the implementation. I appreciate that!

closed 2020-10-02 by shiirish