nanopb/nanopb #598: Field Info does not fit static assert

posipunk, 2020-10-05:
I moved from 0.3.9.2 to the 0.4.3 code and what was working fine before is now a mess of static assertion failures.
I created a very simple proto file to demonstrate the issue and see what was happening. 
My compiler: Segger Embedder Studio 4.16 on Windows x64

The static assertion thrown is:
`variably modified 'pb_static_assertion_FIELDINFO_DOES_NOT_FIT_width1_field1_9_0' at file scope`
This is tied to the PB_BIND line in test.pb.c
I've been using compiler options (PB_NO_ERRMSG, PB_WITHOUT_64BIT, PB_BUFFER_ONLY, PB_FIELD_16BIT) however, running with no options doesn't solve the issue. 
**Defining PB_FIELD_32BIT makes the assertion go away - but clearly shouldn't be needed.**
In my real proto file I do have arrays but none of them allow for more than 255 elements and none of the fields have value > 65535.

test.proto
```
syntax = "proto3";
package test_issue;

// Used to start stimulation
message test_the_field_width_issue{
  uint32 just_one_field = 1;
}

```

Command to generate the proto file:
`nanopb_generator --output-dir=.\my_path\proto test.proto`

Generated test.pb.c
```
/* Automatically generated nanopb constant definitions */
/* Generated by nanopb-0.4.3 */

#include "test.pb.h"
#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif

PB_BIND(test_issue_test_the_field_width_issue, test_issue_test_the_field_width_issue, AUTO)

```

Generated test.pb.h
```
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.3 */

#ifndef PB_TEST_ISSUE_TEST_PB_H_INCLUDED
#define PB_TEST_ISSUE_TEST_PB_H_INCLUDED
#include <pb.h>

#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif

/* Struct definitions */
typedef struct _test_issue_test_the_field_width_issue {
    uint32_t just_one_field;
} test_issue_test_the_field_width_issue;

#ifdef __cplusplus
extern "C" {
#endif

/* Initializer values for message structs */
#define test_issue_test_the_field_width_issue_init_default {0}
#define test_issue_test_the_field_width_issue_init_zero {0}

/* Field tags (for use in manual encoding/decoding) */
#define test_issue_test_the_field_width_issue_just_one_field_tag 1

/* Struct field encoding specification for nanopb */
#define test_issue_test_the_field_width_issue_FIELDLIST(X, a) \
X(a, STATIC,   SINGULAR, UINT32,   just_one_field,    1)
#define test_issue_test_the_field_width_issue_CALLBACK NULL
#define test_issue_test_the_field_width_issue_DEFAULT NULL

extern const pb_msgdesc_t test_issue_test_the_field_width_issue_msg;

/* Defines for backwards compatibility with code written before nanopb-0.4.0 */
#define test_issue_test_the_field_width_issue_fields &test_issue_test_the_field_width_issue_msg

/* Maximum encoded size of messages (where known) */
#define test_issue_test_the_field_width_issue_size 6

#ifdef __cplusplus
} /* extern "C" */
#endif

#endif
```

PetteriAimonen, 2020-10-06:
Hmm, this sounds like the static asserts are not working correctly on that compiler for some reason.

Defining `PB_NO_STATIC_ASSERT` would disable those checks, but is clearly not an optimal solution.

If the compiler options allow specifying standard version, you could set that to C11 or newer because the new C standards have much better support for static assertions than the old 1989 version.

posipunk, 2020-10-08:
Thanks! Hadn't thought to look at that. My C standard was gnu99 (essentially c99), bumping it up to gnu11 (c11) fixed the issue, no compiler warnings.

closed 2020-10-08 by posipunk

PetteriAimonen, 2020-10-09:
Yeah. It's somewhat strange that it seems to work ok with plain GCC even with other standard versions.