nanopb/nanopb #599: Intermittent build failure with "AttributeError: module 'proto.nanopb_pb2' has no attribute 'IS_8'"

BeckmaR, 2020-10-06:
We use nanopb in a project where we also use https://github.com/zephyrproject-rtos/zephyr. The unit test stage randomly fails building our .protos due to the error mentioned in the title, full stack trace below:
```
CMake Error at ../../../../../../CMakeLists.txt:12 (message):
  result=1;stderr=Traceback (most recent call last):

    File "<...>workspace/modules/lib/nanopb/generator/nanopb_generator.py", line 119, in <module>
      (FieldD.TYPE_INT32,   nanopb_pb2.IS_8):   ('int8_t',   'INT32', 10,  1),

  AttributeError: module 'proto.nanopb_pb2' has no attribute 'IS_8'

  --nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```
Zephyr's unit test framework `sanitycheck` builds for multiple architectures, using multiple parallel jobs. Therefore it is very much possible that this error somehow stems from multiple build jobs attempting to build the same .proto-files at once (not to the same target location though).

If there is any way or method to stop this from happening and randomly failing our builds or if you have any pointers for us what to look for, we'd be very glad.

Versions:
nanopb 4.2.0
protoc 3.13.0

PetteriAimonen, 2020-10-06:
That sounds like the file `nanopb_pb2.py` in `generator/proto` directory may be out-of-date. It is automatically generated using `protoc`.

The code in `generator/proto/__init__.py` should automatically update the file, but it relies on timestamps to detect when it needs to be regenerated.

Maybe there is something switching between git branches and then the timestamps somehow don't get updated? One could try to run `git clean -dxf` or similar to remove all autogenerated files before build.

(As an aside, it would be nice if the file `nanopb_pb2.py` could be autogenerated and loaded into Python without ever writing it to filesystem, but I'm not sure if that is possible with python-protobuf library.)

BeckmaR, 2020-10-06:
We also build in a docker container, so the build environment should always be 100% clean.

Do I understand correctly that the file that is auto-generated is located in the actual nanopb directory, not somewhere in the build directory?

In that case, I assume parallel builds could fail in our case because perhaps one process already touched the file, but did not yet write it completely, so another process assumes its fresh and reads a half-written file?

PetteriAimonen, 2020-10-06:
That's possible, though I'd expect a partially written file to cause some kind of syntax error instead.

I now notice that your build is using CMakeLists.txt. Pull request #555 and #556 made some changes to how the CMake build works, and installs the temporary file `nanopb_pb2.py` under the build directory instead. But that is only included in nanopb-0.4.3 and newer.

BeckmaR, 2020-10-07:
I will close this issue since I assume we found the reason for our problem. Thank you for the support @PetteriAimonen!

closed 2020-10-07 by BeckmaR