nanopb/nanopb #601: add support for specifying destination directory for generated files

M1cha, 2020-10-06:
Right now it's expected that the user adds `CMAKE_CURRENT_BINARY_DIR` to the include path which I find kinda hacky and error prone.

There should be an argument to `NANOPB_GENERATE_CPP` to change that directory. If you don't see any side-effects or conceptional conflicts that should be a very simple change and I can even create a PR.

PetteriAimonen, 2020-10-06:
The problem isn't only for CMake, so I think it might be worth it to just get rid of the generated nanopb_pb2.py altogether.

I think it should be possible by calling `protoc --descriptor_set_out=` and passing the result to https://googleapis.dev/python/protobuf/latest/google/protobuf/message_factory.html

PetteriAimonen, 2020-11-25:
Actually duplicate of #490. The branch dev_dynamic_nanopb_pb2 seems to work ok, though it is a bit wasteful when it recompiles the file every time the generator is run. It would be nice if there was a smarter way to cache dynamic python modules.