nanopb/nanopb #603: nanopb_generator does not report a version number

mrkita, 2020-10-08:
I can't find any sort of version number reporter for the python generator (at least in the compiled binaries provided in the website).

This would be very helpful for provisioning systems, that can use that to verify the installed version number.

PetteriAimonen, 2020-10-08:
Yeah, it would be useful to add a `--version` switch.

In the meanwhile, one can use this snippet: `python -c 'import nanopb_generator; print(nanopb_generator.nanopb_version)'`

mrkita, 2020-10-08:
Thanks that's a good suggestion. The problem is, we don't have python installed directly, nor do we want to include it as a dependency (nothing else in our project makes use of it).

tknerr, 2020-10-08:
+1, a `--version` flag would definitely be helpful 👍 :)

closed 2020-11-25 by PetteriAimonen