nanopb/nanopb #604: Linking error with conan package

a1lu, 2020-10-09:
Hi,
I linked against the nanopb conan package and got the following errors:

```
undefined reference to `pb_ostream_from_buffer'
undefined reference to `pb_encode'
undefined reference to `pb_istream_from_buffer'
undefined reference to `pb_decode'
```
I noticed that the conan recipe does not specify the library to link against, so I added
```
    def package_info(self):
        self.cpp_info.libs = ["protobuf-nanopb"]
```
to conanfile.py and linking works again.

PetteriAimonen, 2020-10-09:
@amatosov-rbx, @Psy-Kai Do you want to take a look?

Psy-Kai, 2020-10-09:
I strumbled upon this problem myselfe a few days ago but hat no time to create a pull request. Yes the cpp_info.libs is missing.

PR #605 (2020-10-10): Add protobuf-nanopb library to cpp_info.libs
Without this, there will be no links against the nanopb library when
nanopb is integrated by conan
Fixes #604
closed 2020-10-10 by PetteriAimonen