nanopb/nanopb #606: Can't generate stubs on Windows

gehreleth, 2020-10-10:
**Starting conditions**: 

Nanopb release:

C:\..\nanopb>git checkout tags/0.4.3
HEAD is now at 1466e6f Publishing nanopb-0.4.3

C:\..\nanopb>git branch
* (HEAD detached at nanopb-0.4.3)
  master

Visual Studio Community Edition 2019, Built-in Python, have protobuf and grpcio-tools in python

>python.exe
Python 3.7.8 (tags/v3.7.8:4b47a5b6ba, Jun 28 2020, 08:53:46) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.

>pip install protobuf grpcio-tools
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: protobuf in c:\users\serge\appdata\roaming\python\python37\site-packages (3.6.0)
Requirement already satisfied: grpcio-tools in c:\users\serge\appdata\roaming\python\python37\site-packages (1.32.0)
Requirement already satisfied: setuptools in c:\program files (x86)\microsoft visual studio\shared\python37_64\lib\site-packages (from protobuf) (47.1.0)
Requirement already satisfied: six>=1.9 in c:\users\serge\appdata\roaming\python\python37\site-packages (from protobuf) (1.15.0)
Requirement already satisfied: grpcio>=1.32.0 in c:\users\serge\appdata\roaming\python\python37\site-packages (from grpcio-tools) (1.32.0)

I've got a .proto file that's being successfully compiled in Linux without any issues, tweak.proto

**What I'm trying to achieve**:

I don't want to include autogenerated files inside release, thus my goal is to write cmake that's generating stubs during build process. Since cmake task relies on python scripts from _generator_ directory, I'm trying to make them work as expected. 

Process: 
 - I've copied tweak.proto to nanopb directory
 - I run >.\generator\protoc.bat tweak.proto

**Expectation**: 

Appearance of tweak.pb.h and tweak.pb.c in the same directory.

**Actual result**:

Traceback (most recent call last):
  File "C:\...\nanopb\generator\nanopb_generator.py", line 51, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\...\nanopb\generator\\protoc", line 6, in <module>
    from nanopb_generator import invoke_protoc
  File "C:\...\nanopb\generator\nanopb_generator.py", line 72, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "C:\...\nanopb\generator\proto\nanopb_pb2.py", line 23, in <module>
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'

PetteriAimonen, 2020-10-11:
> File "C:...\nanopb\generator\proto\nanopb_pb2.py", line 23, in
create_key=_descriptor._internal_create_key,
> AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'

Hmm, just a guess, but try removing `generator/proto/nanopb_pb2.py`, as there seems to be some kind of version mismatch. It should be automatically regenerated.

gehreleth, 2020-10-13:
This file seem to get regenerated indeed, but I'm not sure what versions are you talking about.
Can you please elaborate preconditions for protoc.bat to work on Win10 installation.
I've installed default pip3 suggestions for protobuf and grpcio-tools.
I take that they won't match with ones needed for your code generator, am I correct?

PetteriAimonen, 2020-10-14:
I'm referring to Python versions. The pip3 suggestions for protobuf and grpcio-tools should work fine. But sometimes there is a problem that the system has several versions of python-protobuf or Python itself installed, and they get mixed up. Unfortunately these are quite annoying to debug.

closed 2020-11-25 by PetteriAimonen

gksrb4, 2021-04-07:
It works for me to upgrading python version.