nanopb/nanopb #609: Type error compiling for C8051

EmbeddedDesign, 2020-10-29:
Using the latest stable nanopb-0.4.3 attempting to compile for a C8051 based SiLabs EFM8BB2 with Keil 8051 v9.60.0

`bool found;` in struct pb_extension_s in pb.h produces error: **bit member in struct/union**

I suspect this is due to the fact that stdbool.h for this system defines bool as a [bit](https://www.keil.com/support/man/docs/c51/c51_le_datatypes.htm).

`pb_decode_tag()` and `pb_decode_bool()` in pb_decode.h also seem to have an issues with using a bit as a pointer in their `bool *eof` and `bool *dest` arguments respectively.

Changing the expected types for these arguments to something that's not a bit does seem to resolve these errors but then produces a **missing ';' before 'pb_type_t'** error for the line `typedef uint_least8_t pb_type_t;` in pb.h, as well as bunch of **'NULL' could not be resolved** errors in pb_decode.c though I'm not yet certain whether these issues are related to the bool being defined as a bit problem at hand or not.

PetteriAimonen, 2020-10-29:
The latter error could be if the compiler doesn't define `uint_least8_t`.

EmbeddedDesign, 2020-10-29:
> The latter error could be if the compiler doesn't define `uint_least8_t`.

Thanks @PetteriAimonen the lack of a definition for `uint_least8_t` was indeed initially an issue.

I added a `#define PB_SYSTEM_HEADER` as [suggested in pb.h](https://github.com/nanopb/nanopb/blob/9f57cc871d8a025039019c2d2fde217591f4e30d/pb.h#L69) to utilize [pb_syshdr.h](https://github.com/nanopb/nanopb/blob/master/extra/pb_syshdr.h) with any occurrences of redefinitions of existing system types and functions commented out.

Oddly, the `#define NULL 0` is definitely there but this error only pops up after trying to tweek a bunch of things to resolve the bit bool issue so I'm hoping that if I can solve that this latter issue won't persist.

PetteriAimonen, 2020-10-29:
I guess the problem with 1-bit `bool` is one more point in favor of adding `pb_bool_t` in issue #287.

closed 2020-11-25 by PetteriAimonen