nanopb/nanopb #610: Error "'dict_values' object does not support indexing" while compiling .proto files

Psy-Kai, 2020-11-09:
It seems like some python2 syntax is in `nanopb_generator.py`:

https://github.com/nanopb/nanopb/blob/9f57cc871d8a025039019c2d2fde217591f4e30d/generator/nanopb_generator.py#L990

This will lead to the error in the title.

PetteriAimonen, 2020-11-09:
Yeah, looks like this code path is uncovered by current test cases.

Psy-Kai, 2020-11-11:
It seems that there is an even bigger issue around this line. Then "fixing" this python2 syntax line via `list(dynamic_sizes.values())[0]` non compilable code will be generated.

PetteriAimonen, 2020-11-11:
Do you have a .proto file that triggers this?

lopsided98, 2020-11-19:
I can reproduce this with the following set of proto files:
```protobuf
syntax = "proto3";

import "LogMessage.proto";

message DataPacket {
    oneof packet {
        LogMessage log_message = 1;
    }
}
```
```protobuf
syntax = "proto3";

message LogMessage {
    string msg = 1;
}
```

The error only occurs if the messages are defined in different files.

PetteriAimonen, 2020-11-20:
Hmm yeah, and it even occurs when both files are passed to the generator together. I'll try to find time to take a closer look at this.

PetteriAimonen, 2020-11-25:
@Psy-Kai Do you have a .proto file that reproduces the problem with **"Then "fixing" this python2 syntax line via list(dynamic_sizes.values())[0] non compilable code will be generated."**? For the file posted by @lopsided98 that fix appears to work correctly.

Psy-Kai, 2020-11-26:
Sorry of the very late answer. I will attach some files which represents the bug (which still exists if you have a deeper structure)
[nanopb_generator_bug.zip](https://github.com/nanopb/nanopb/files/5601989/nanopb_generator_bug.zip)

PetteriAimonen, 2020-11-26:
@Psy-Kai Thanks, I was able to reproduce that bug now. Looks like it has a problem with finding the dependencies of the dependencies.

Psy-Kai, 2021-02-08:
@PetteriAimonen I just tested generating files with generator from master and it works for my proto definitions now. Thanks :)

This also fixed an size-macro ordering issue I got now ^^

PetteriAimonen, 2021-02-08:
Glad to hear that :)

PetteriAimonen, 2021-03-22:
Fix released in 0.4.5.

closed 2021-03-22 by PetteriAimonen