nanopb/nanopb #611: Enum messages inside OneOf generated as ENUM instead of UENUM

awolokita, 2020-11-10:
The generated field type for an enum submessage inside a OneOf is a ENUM, while the same enum not inside a OneOf is generated as a UENUM. This leads to enum values with the MSB set being encoded as a negative number.

Example:
```
enum my_enum {
  E0 = 0;
  E240 = 240;
}

message my_message_regular {
   required my_enum MyEnum = 1;
}

message my_message_oneof {
  required uint32 Data1 = 1;
  oneof Data {
    bool null = 2;
    my_enum MyEnum = 3;
  }
}
```

Results in the my_message_oneof.Data.MyEnum field being generated as ENUM and MyEnum.E240 is decoded as -16 with other protobuf decoders. However, my_message_regular.MyEnum is generated with a type of UENUM as is correctly decoded as an unsigned integer.

PetteriAimonen, 2020-11-10:
Thanks for the bug report.

As a temporary workaround, one can add dummy entry like `Edummy = -1` to force the compiler to allocate 16 bits for the enum type and avoid the signed vs. unsigned confusion.

awolokita, 2020-11-10:
Thanks @PetteriAimonen, the workaround works when decoding with protobufjs.

closed 2020-11-25 by PetteriAimonen