nanopb/nanopb #617: Unordered field numbers inside oneof can cause fields to be ignored

PetteriAimonen, 2020-11-25:
Given a following message type:

     message Foo {
         oneof foo {
              int32 x1 = 1;
              int32 x2 = 200;
         }
         int32 x3 = 2;
     }

can cause field x3 to be ignored in decoding if it occurs after x2 in encoded format. Typically encoders order fields by tag number and this doesn't happen, but it can happen with nanopb's own encoder and probably others.

The bug is present in versions 0.4.3 and 0.4.4, it was introduced in commit 48e875de.

zacwbond, 2021-03-18:
I have a message type defined this way and I am experiencing a slightly different problem in the latest code.  Using your example, I am creating a message that looks like this:

```
Foo foo;
foo.which_payload = Foo_x2_tag
foo.x2 = 0;
foo.x3 = 75;
```

Upon decoding, I find that `foo.which_payload` is 0 instead of 200.  This worked with version 0.4.1.  I can't tell when it broke because 0.4.2 and 0.4.3 generate invalid code for some of my other (unrelated) schema.  It does not work in 0.4.4 or in master.

For reference, here is a simplified version of my PB schema.  The only value that fails to decode properly is the "NoDecode" type; all the rest decode without a problem.

```

message DecodesOK 
{
    uint32          a   = 1;
    uint32          b   = 2;
}

message NoDecode
{
    string          name = 1;
}

message TestMessage
{
    enum MessageType
    {
        A   = 0;
        B   = 1;
        C   = 2;
    }

    MessageType     messageType = 1;
    uint32          x           = 2;
    uint32          y           = 3;
    uint32          ip1         = 4;

    oneof payload
    {
        DecodesOK       pla5    = 5;
        DecodesOK       pla6    = 6;
        DecodesOK       pla7    = 7;
        DecodesOK       pla8    = 8;
        DecodesOK       pla9    = 9;
        DecodesOK       pla10   = 10;
        NoDecode        plb11   = 12;
    }

    uint32          ip2         = 11;
}
```

Here's the options file:

```
* anonymous_oneof: true
* long_names     : false

NoDecode.name   max_length     : 32
```

Here's the encoded hex that nanopb is not decoding correctly, but which does include a default field 12.  

`0801100F18AC0220F182A0850C620058F182A0850C`

An [online tool](https://protogen.marcgravell.com/decode/) decodes it correctly, but nanopb sets `which_payload` to 0 instead of 12.

```
Field #1: 08 Varint Value = 1, Hex = 01
Field #2: 10 Varint Value = 15, Hex = 0F
Field #3: 18 Varint Value = 300, Hex = AC-02
Field #4: 20 Varint Value = 3232235889, Hex = F1-82-A0-85-0C
Field #12: 62 String Length = 0, Hex = 00, UTF8 = ""
Field #11: 58 Varint Value = 3232235889, Hex = F1-82-A0-85-0C
```

PetteriAimonen, 2021-03-19:
Thanks for reporting, it is a bug indeed. Closely related to #617, but the fix for that is apparently incorrect / incomplete.

Part of the issue is that fields inside oneof are not getting correctly ordered in the generated array in `.pb.h`:

<pre>
...
X(a, STATIC,   ONEOF,    MESSAGE,  (payload,pla10,payload.pla10),  10) \
X(a, STATIC,   ONEOF,    MESSAGE,  (payload,plb11,payload.plb11),  12) \
X(a, STATIC,   SINGULAR, UINT32,   ip2,              11)
</pre>

Which ends up triggering this line here: https://github.com/nanopb/nanopb/blob/master/pb_common.c#L215

As a workaround, one could comment out line 215 of pb_common.c, as it is only a speed optimization.

The real fix would be to make the generator properly sort the fields. In 0.3.x all oneof fields had to be next to each other because of the descriptor format, but 0.4.x removed that restriction.

PetteriAimonen, 2021-03-22:
Fix released in 0.4.5.

closed 2021-03-22 by PetteriAimonen